LibreCAD/LibreCAD #817: Cannot enter rlative coordinates when command window is detached

Zmuogs, 2016-09-17:
I use LibreCad with detached command window.
 I can enter absolute coordinates (digit symbols, coma, point, slash) while drawing of entity is in progress. But I can't enter relative coordinates: when I want to enter a @ symbol in to command prompt then I press two keys [Shift] and [2,@], but in command prompt mentioned symbol not appears. Conflict with "Shift" drawing mode?

Version: 2.2.0-alpha
Compiler: GNU GCC 5.4.0
Compiled on: Sep 15 2016
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Ubuntu 16.04.1 LTS

r-a-v-a-s, 2016-09-17:
Actually it's because of the patch for your other command line issue. #790
Somehow in the two months since that patch,
no one reported this... until today, the day after I released 2.1.2,
which included the patch... 😲 

I think I've fixed the problem, but for now you can try the new x..y syntax.
In theory it's more efficient, and allows for using the numpad exclusively.

r-a-v-a-s, 2016-09-17:
Users of 2.1.2 can press Spacebar before Shift if the command line isn't focused.
The space character will be ignored.
To solve this, 2.1.3 will be released on the equinox (22nd).

Zmuogs, 2016-09-17:
Thanks.
 This bug report is caused by a fresh nightly builds which are now generated again after looong pause :)

r-a-v-a-s, 2016-09-17:
Apparently my post got the maintainer's attention.
https://answers.launchpad.net/ubuntu/+source/librecad/+question/395705

Enjoy!

r-a-v-a-s, 2016-09-21:
I'm requesting that testers try all the ways they use modifier keys.
Everything I could think of works with my latest patch.

The command line hijacks keyboard input for commands,
so we need to filter the modifier keys.
Two complications are:

1. shift is used in the drawing area, but also needs to activate the command line for `@`
2. it seems like numpad input is considered modifier input (`Qt::KeypadModifier`)

closed 2016-09-23 by r-a-v-a-s