LibreCAD/LibreCAD #818: Incorrect "trim two" when line and arc is involved

Zmuogs, 2016-09-17:
Try function "Trim two" on attached test drawing "Untitled2.dxf" (view in "inintial.png"). Drawing contains two Line segments and an Arc, ends of which touches line segments.
 Now try to use "trim two":
 - if you click on left Line segment and left side of Arc then you will get incorrect result: "trim_two_left.png";
 - if you click on right Line segment (on far right side of line) and right side of Arc then you will get another incorrect result: "trim_two_right.png".
 Correct result should be a "cross section of a hat".

Version: 2.2.0-alpha
Compiler: GNU GCC 5.4.0
Compiled on: Sep 15 2016
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Ubuntu 16.04.1 LTS

Zmuogs, 2016-09-17:
[image]
[image]
[image]

lordofbikes, 2022-07-24:
Found this by investigating #1555.

What you recognized here is intended behavior.
Basically the trim tools are affine to click positions. That means you have control over the trim result with the click positions. It depends if you click near or far of an intersection. This will advice the trim tool to use different strategies.
So to get the hat cross section both entities, the arc and the left line has to be clicked near to their intersection.

Even when the cases you titled wrong, don't make sense in this example, they may be useful in other constellations or with other entities.

Basically my comment in #1555 applies here too. When you know that trim will work go with it, but in vague cases use divide and delete unneeded parts.

closed 2022-07-24 by lordofbikes