LibreCAD/LibreCAD #819: "Explode" messing with colors of rectangle

Zmuogs, 2016-09-17:
Steps to reproduce:
 - start a new drawing,
 - create second layer,
 - color of second layer change to Green,
 - while second layer is "current" draw a Rectangle:[image]
 - activate command "Explode" and explode mentioned Rectangle: you will get lines with various colors:[image]

Version: 2.2.0-alpha
Compiler: GNU GCC 5.4.0
Compiled on: Sep 15 2016
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Ubuntu 16.04.1 LTS

lordofbikes, 2016-09-17:
I can't test it at the moment, but I think the problem isn't the color. My assumption is, that the white lines are created on layer 0, which default color is white.
If so, the problem is related to losing the layer information for the new generated entities.
Can you confirm this?

Zmuogs, 2016-09-18:
I used "Modify" -> "Properties" on Lines: Layer stays same, Color changes:
[image]
[image]

Equoda, 2016-09-18:
Yes, I confirm this, but can't investigate now due to the debugger problems.
This may be related to #755 because a rectangle is a some kind of container, like a block.

FelixNotTheCat, 2017-01-14:
@lordofbikes @r-a-v-a-s For RS2::EntityPolylines, RS_Modification::explode() (in rs_modification.cpp) resolves layers (resolveLayer = true;) but not colours (resolvePen = false;). resolvePen = true; then resolves colours of all entities in a (polylines) container. Pls say whether I can commit the change.
(PS the containers first entity gets its colour from the container as opposed to from the preceding entity in the loop.)

closed 2019-04-19 by lordofbikes