LibreCAD/LibreCAD #820: two rows of block icons

r-a-v-a-s, 2016-09-19:
The block actions are in the list's right-click context menu;
and we can add the actions to custom toolbars and menus. 
Therefore, I propose reclaiming some of the valuable vertical space for lists.

Some ideas:

- eliminate the icons in favor of the context menu / custom widgets
- transform the rows into columns (astride the list)
- reduce to one row or column

Do you think any of the icons are essential as part of the dockwidget?

fa201, 2020-04-14:
I like the way it is now. Since nobody else replied, should we keep this issue opened ?

Symbian9, 2021-05-21:
> Since nobody else replied, should we keep this issue opened ?

Guess, yes.

Next release of QCAD (`3.26.3` or `3.27`) would include new feature, which is some sort of mentioned in this thread:

- https://qcad.org/en/changelog
> - **QCAD Professional:**
>   - Edit > Application Preferences > Tool Settings:
>     - Add option to add tools to user defined toolbars