LibreCAD/LibreCAD #822: the system for tool plugins

r-a-v-a-s, 2016-09-23:
Currently, tool plugins make use of `getEnt()`, `getPoint()` and `getSelect()`,
in `doc_plugin_interface`, which isn't actually a Qt plugin interface (i.e. `qc_plugininterface`).
These functions use a loop that I guess is concurrent with LibreCAD's main loop.

```c++
        QEventLoop ev;
        while (!a->isCompleted()) {
            ev.processEvents ();
            if (!gView->getEventHandler()->hasAction())
                break;
        }
```

This approach has produced several bizarre bugs, and seems forced.

If I was inspired to do the work I would make a Qt pugin interface that mirrors rs_actioninterface;
and I would modify rs_eventhandler to accept 'native' and 'plugin' tools with a state for each.
Although, there's probably more to consider...

Xia-Weiwen, 2016-12-19:
Hi, @r-a-v-a-s Do you mean that plugin developers could define customized actions if rs_actioninterface were made a plugin interface? I've been studying the plugin codes these days.

r-a-v-a-s, 2016-12-19:
Yes, but I was imagining making a copy of rs_actioninterface.

I'm no longer working on LibreCAD,
which means there are no inspired developers currently;
and that means [LibreCAD 3][1] will be the more active project from now on.

[1]:https://github.com/LibreCAD/LibreCAD_3

Xia-Weiwen, 2016-12-20:
I see. Thank you !

r-a-v-a-s, 2016-12-22:
LC3 is a completely different program, 
where the core will not be copyleft software,
and will not be dependent on Qt.
It already has support for lua scripting;
and I've been told network collaboration is an interest.

http://forum.librecad.org/LibreCAD-Engine-td5712298.html
http://forum.librecad.org/plugin-interface-for-LibreCAD-3-td5714404.html

The founder of LibreCAD stopped working on LC2 and started LC3.
dxli went into the private sector.
Rallaz is too busy with work and family these days.
I don't need CAD now, and have a lot of other interests.

There is a lot of work that needs to be done on LC2, 
e.g. the plugin system, rewriting the intersection detection system,
cleaning / rewriting of the drawing system, optimization for splines,
support for regular fonts, support for scripting, various bugs and inconsistencies...
and the list goes on.
This project doesn't currently have a leader in the sense of a visionary;
and it doesn't seem to have anyone interested in documentation either.
New users / contributors will be receiving much less advice / support because of absent authors.

A modern Qt based CAD would ideally use the Graphic View Framework,
and support javascript via QJSEngine.
https://github.com/GreatDevelopers/eCAD

You can draw your own conclusion from that. ;-]

rvt, 2016-12-22:
Xia,

I am the person that started the fork of QCad into LibreCAD 2 and I started working on LibreCAD 3. Although I don't see myself as the founder, the community picked the LibreCAD name and the community designed the new iconset, logo and other artwork.

LibreCAD 2 (our current version) will not be abandoned and we will keep progressing it at different rates. Ravas has done great work supporting us on LibreCAD 2 but unfortunately is going to work on other areas of his interest. Remember, this is open source, this happens, everybody is free to come and go whenever they want, we cannot control that nor I want to control that!

_and it doesn't seem to have anyone interested in documentation either._
This is one of the harder parts, nobody does seem to have interested in a documentation, although our wiki has a lot of documentation and there are youtube movies that show LibreCAD.

_This project doesn't currently have a leader in the sense of a visionary;_
* I might be biassed saying this but we do have a vision for LibreCAD (a shared vision of all current developers), this is why we are here today compared to 6 years ago. This is why we participate in Google Code In and Google Summer of Code, to get new people interested in coding and CAD. We hope that a few people stick around and help us our, like Florian who participated in Google Summer of Code and is now a mentor at Google Code In. LordOfBikes who is a long term supporter and also a mentor at Google Code In. 
Our developers that work on LibreCAD often have there own vision, perhaps they like coding or perhaps they like to try things out. If it helps the project, we accept the code, we have been VERY liberal in that!

_A modern Qt based CAD would ideally use the Graphic View Framework, and support javascript via QJSEngine._
- This isn't true, modern CAD engines should be pluggable such that a user/developer is free to choose it's own UI framework, or any other framework for that matter.
- A CAD application doesn't have to be Qt based, there are excellent examples of applications that are not Qt based and do have very nice frameworks.
- The Graphic View Framework is nice, but doesn't solve all issues related to CAD, and for a very long time this had issue with accuracy's at higher zoomlevels, at least drawftsight also had issues with that.
- I am not sure nowdays, but Graphic View Framework used a BSP (Binary Space partitioning) and as far as I know isn't pluggable, that means if we want to use something else more suitable for CAD applications we cannot change it.

_Our vision for LibreCAD 3_
With LibreCAD 3 we are trying to create a framework/libraries set such that other applications can use our rendering, storage and manipulation 'engine' (I don't have a better word for engine). For example BRL-CAD would love to use LibreCAD's capabilities, however they are not Qt based and thus cannot use LibreCAD 2 (also due to license issues). With LibreCAD 3 we are trying to solve this with a more liberal license and better application design (this is always debatable!).

Hopefully that answers some questions and/or clarify some items.

R. van Twisk

r-a-v-a-s, 2016-12-22:
> I might be biassed saying this but we do have a vision for LibreCAD

This exemplifies the problem I've pointed out about the name "LibreCAD 3".
My statement (that whole paragraph) was about LC2, 
which does not have a leader or visionary.

> This isn't true, modern CAD engines should be pluggable such that a user/developer is free to choose it's own UI framework, or any other framework for that matter.

I said "A modern Qt based CAD"
LC3 isn't a Qt based CAD... So again I wasn't talking about that.
Regarding the use of the word "should": that's your opinion.
I was talking about a project where people want to fully use Qt.
My point is that LC2 is completely dependent on Qt at this point,
and could be using the Graphic View Framework,
which would improve drawing performance 
and would allow for regular fonts, among other things.
Just because something has a bug at one point isn't a reason to never use it,
it's a reason to fill out a bug report.

Xia-Weiwen, 2016-12-24:
Thank both of you very much for the explanation!
I would like to start from LC2 and keep learning.
Merry Christmas!

fa201, 2020-04-13:
Is this "issue" still valid, I mean the original one related to plug-in ?

sand1024, 2026-09-26:
outdated

closed 2026-09-26 by sand1024