LibreCAD/LibreCAD #823: Mirror of blocks causes them to be offset

MPForsyth, 2016-10-01:
On mirroring some blocks (in this case 2x4 blocks from lower to upper wall) using the midpoint of lines LibreCAD shows them in the correct position until I klick on the second mirror point.  After the dialog box
[chicken-house.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/504695/chicken-house.dxf.zip)

 they are offset.
I tried to duplicate on a simple drawing but was not able to do so.
Attached is the drawing where it happens. 
Version: master
SCM Revision: 2.0.8
Compiler: GNU GCC 5.2.1
Qt Version: 4.8.6
Compiled on: Mar 16 2016

MPForsyth, 2016-10-01:
Using the same drawing at a later stage.  Mirroring a wall section, the blocks are again offset from the wall lines by what looks like the height of the block.  New file attached
[chicken-house2.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/504710/chicken-house2.dxf.zip)

lordofbikes, 2016-10-08:
I can confirm this is a bug for now

lordofbikes, 2016-10-09:
I've found the source for this bug. It's a bit tricky and I'm not sure if I can provide a solution quickly.

Mirroring works generally for inserts. But your block `2x4r` is a rotated version of block `2x4`.
And this is, where the bug comes in. Mirroring a block inside a block fails.

For your drawing you can easily work around by modifying block `2x4r`. Delete the inserted block `2x4` in `2x4r` and draw the rect in the block by 4 single lines or a rectangle. Then mirroring the `2x4r` block should work too.

MPForsyth, 2016-10-09:
Ok, I can live with that.  Thanks for the answer

lordofbikes, 2016-10-10:
I reopen it to keep it in mind.

fvigano, 2019-10-10:
There is another problem related to mirroring of blocks: when you try to mirror blocks containing text, the text does not mirror, but it does a sort of rotation. Please, find attached the screenshots of before and after rotation of the block (the selected rotation axis is the green line).
Anyway, a great piece of sofware.

[image]
[image]

fa201, 2020-04-13:
I cannot reproduce this. The test in the block is not mirrored but disappeared. Switching ON/OFF Draft mode did not change. Saving, closing and re-opening did not change either.
[image]

If I explode the block it is the same but the text has a negative value.
[image]

Removing the negative sign reproduce fvigano issue.
[image]

Recent build:
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

sepastian, 2022-04-07:
Hello, thanks for LibreCAD, its awesome!

Any update on this? The problem still exists, mirroring results in unexpected results. I need to mirror blocks (to construct a hilbert curve).