LibreCAD/LibreCAD #826: Issue hatching areas with nested shapes

kostic2000, 2016-10-03:
When trying to hatch a shape that has another enclosed shape, in some cases the fill is incorrect. It seems that some sharp edges make a difference, but not 100% sure.
Attached is an example DXF, where the areas are letters of a particular font converted to curves (by doing Explode).

Original drawing:
[image]

Select all curves of one letter at a time and do a solid hatch. Here is what I get:
[image]

And here what I expected to get:
[image]

Source DXF supplied in the zip
[hatch_issue.zip](https://github.com/LibreCAD/LibreCAD/files/507276/hatch_issue.zip)

r-a-v-a-s, 2016-10-31:
It's reproducible simply by creating a rectangle inside another rectangle.

FelixNotTheCat, 2017-01-12:
optimizeContours() isn't evolved enough yet to cope with selected, intersecting entities. ravas, could you please assign me (-assuming that's the way to do it) and i try beef it up

lordofbikes, 2017-01-12:
@FelixNotTheCat you're welcome to work on this, but assigning is prohibited to team members.
So I assigned myself to give you the chance to work on it.
The recommend way is to fork LibreCAD and work on your own fork. Then we can discuss solutions there and finally create a pull request that we can merge your solution.

If you hit any issues, don't hesitate to ask here. Thanks for your contribution.

FelixNotTheCat, 2017-01-14:
@lordofbikes many thanks Armin -simply don't want to step on others' toes, or do redundant work. Pls see today's LibreCAD/ commit (i.e. rs_hatch.cpp and rs_entitycontainer.cpp). It's far from a complete (hatch) solution but addresses issues such as highlighted by kostic2000; your/LC team's critiscm/comments/feedback etc is much appreciated.

commit 5480eba6 (2022-04-16): fixed condition for solid fill hatch polygones, fixes #826,#1353,#1354
closed 2022-04-16 by lordofbikes