LibreCAD/LibreCAD #827: Centroid function

sven98de, 2016-10-07:
It would be nice to have a tool to determine the centroid of polygons.
Are there any plans to implement it in future releases, perhaps as a plugin ?

Sven

r-a-v-a-s, 2016-10-13:
I'm in maintenance mode, and I'm not aware of any other developers working on new features.

This seems useful enough to be a core feature.

fa201, 2023-08-09:
There is no tool implemented but the work around is very simple: draw a circle passing through 3 points of polygon. Then there is a center.

I suggest we close this issue.

sven98de, 2023-08-10:
Really, what's the relation between a center of a circle drawn through 3 point of a polygon and the center of gravity?

fa201, 2023-08-10:
Understood, I was mislead but the polygon tool in LibreCAD.

sand1024, 2026-09-26:
supported via hatch properties dialog

closed 2026-09-26 by sand1024