LibreCAD/LibreCAD #829: Segfault ,numerous, no particular association

MPForsyth, 2016-10-17:
Version: master
SCM Revision: 2.0.8
Compiler: GNU GCC 5.2.1
Qt Version: 4.8.6
Compiled on: Mar 16 2016
Crashes about every 10 to 20 minutes.  Not associated with any particular operation.  Only program that crashes on this machine.  About 1 out of 5 crashes results in submitting error report.   Most drawings are similar in complexity to demo  L  file.

r-a-v-a-s, 2016-10-17:
That's to be expected with 2.0.8.
The latest version is 2.1.3 (6 versions ahead).

closed 2016-10-17 by r-a-v-a-s

MPForsyth, 2016-10-17:
thanks for the reply,   I was relying on the stable release ppa  which seems to stop at 2.0.8   for wiley. Is that true?
I have now gone to the daily release  and 2.2.0alpha.

r-a-v-a-s, 2016-10-17:
Yes. 2.2.0 is basically ready for release, so it's stable. Let us know if you experience any crashes with that version.

r-a-v-a-s, 2016-10-17:
Note there is one report of a crash using that PPA
http://forum.librecad.org/Bug-Crash-when-click-drag-corner-of-rectangle-to-enlarge-td5714359.html

I'm unable to reproduce it.

r-a-v-a-s, 2016-10-17:
https://github.com/LibreCAD/LibreCAD/issues/830