LibreCAD/LibreCAD #830: crash when moving object directly after loading a DXF

r-a-v-a-s, 2016-10-17:
http://forum.librecad.org/Bug-Crash-when-click-drag-corner-of-rectangle-to-enlarge-td5714359.html
It doesn't seem to matter what type of entity.

1. create an entity
2. save
3. close
4. reopen
5. try to move the entity with the default selection tool

It stops on line 666 (devil's in the details) of stl_tree.h claiming `this` is `0x0`.
Full backtrace:
```

Thread 3 (Thread 0x7fffe96d7700 (LWP 2532)):
#0  0x00007ffff52206ed in poll () at /lib64/libc.so.6
#1  0x00007ffff4a12016 in  () at /usr/lib64/libglib-2.0.so.0
#2  0x00007ffff4a1212c in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#3  0x00007ffff64f7aab in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#4  0x00007ffff649f6fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#5  0x00007ffff62c53c3 in QThread::exec() () at /usr/lib64/libQt5Core.so.5
#6  0x00007fffec1c2985 in  () at /usr/lib64/libQt5DBus.so.5
#7  0x00007ffff62ca2e8 in  () at /usr/lib64/libQt5Core.so.5
#8  0x00007ffff5d8a454 in start_thread () at /lib64/libpthread.so.0
#9  0x00007ffff52293ff in clone () at /lib64/libc.so.6

Thread 2 (Thread 0x7fffead0e700 (LWP 2531)):
#0  0x00007ffff52206ed in poll () at /lib64/libc.so.6
#1  0x00007ffff1c6f3f0 in  () at /usr/lib64/libxcb.so.1
#2  0x00007ffff1c71189 in xcb_wait_for_event () at /usr/lib64/libxcb.so.1
#3  0x00007fffee7987a9 in  () at /usr/lib64/libQt5XcbQpa.so.5
#4  0x00007ffff62ca2e8 in  () at /usr/lib64/libQt5Core.so.5
#5  0x00007ffff5d8a454 in start_thread () at /lib64/libpthread.so.0
#6  0x00007ffff52293ff in clone () at /lib64/libc.so.6

Thread 1 (Thread 0x7ffff7fa7740 (LWP 2527)):
#0  0x000000000052d4f8 in std::_Rb_tree<RS_Undoable*, RS_Undoable*, std::_Identity<RS_Undoable*>, std::less<RS_Undoable*>, std::allocator<RS_Undoable*> >::_M_begin() (this=0x0) at /usr/include/c++/6/bits/stl_tree.h:666
#1  0x00000000005d2f7c in std::_Rb_tree<RS_Undoable*, RS_Undoable*, std::_Identity<RS_Undoable*>, std::less<RS_Undoable*>, std::allocator<RS_Undoable*> >::_M_get_insert_unique_pos(RS_Undoable* const&) (this=0x0, __k=@0x7fffffffc540: 0x1d05780) at /usr/include/c++/6/bits/stl_tree.h:1830
        __x = 0x100000000000000
        __y = 0x4050300000000003
        __comp = false
        __j = {_M_node = 0x0}
#2  0x00000000005d2d9a in std::_Rb_tree<RS_Undoable*, RS_Undoable*, std::_Identity<RS_Undoable*>, std::less<RS_Undoable*>, std::allocator<RS_Undoable*> >::_M_insert_unique<RS_Undoable* const&>(RS_Undoable* const&) (this=0x0, __v=@0x7fffffffc540: 0x1d05780) at /usr/include/c++/6/bits/stl_tree.h:1889
        __res = {first = 0x8e2716, second = 0xee6cf0}
#3  0x00000000005d2ca1 in std::set<RS_Undoable*, std::less<RS_Undoable*>, std::allocator<RS_Undoable*> >::insert(RS_Undoable* const&) (this=0x0, __x=@0x7fffffffc540: 0x1d05780) at /usr/include/c++/6/bits/stl_set.h:483
        __p = {first = {_M_node = 0xfb6d80}, second = 104}
#4  0x00000000005d29ba in RS_UndoCycle::addUndoable(RS_Undoable*) (this=0x0, u=0x1d05780) at lib/engine/rs_undocycle.cpp:9
#5  0x000000000052aff6 in RS_Undo::addUndoable(RS_Undoable*) (this=0x1cdd1c0, u=0x1d05780) at lib/engine/rs_undo.cpp:108
        __func__ = "addUndoable"
#6  0x00000000005b8970 in RS_Modification::deselectOriginals(bool) (this=0x7fffffffc680, remove=true) at lib/modification/rs_modification.cpp:2053
        selected = true
        e = 0x1d05780
        __for_range = @0x1cdd0c0: {<RS_Entity> = {<RS_Undoable> = {<RS_Flags> = {_vptr.RS_Flags = 0xe8cf48 <vtable for RS_Graphic+16>, flags = 2}, cycle = 0x0}, parent = 0x0, minV = {x = 39.25, y = 45.25, z = 0, valid = true}, maxV = {x = 177.5, y = 168.5, z = 0, valid = true}, layer = 0x0, id = 12973, pen = {<RS_Flags> = {_vptr.RS_Flags = 0xe89208 <vtable for RS_Pen+16>, flags = 0}, lineType = RS2::SolidLine, width = RS2::Width00, screenWidth = 0, color = {<QColor> = {cspec = QColor::Rgb, ct = {argb = {alpha = 65535, red = 0, green = 0, blue = 0, pad = 0}, ahsv = {alpha = 65535, hue = 0, saturation = 0, value = 0, pad = 0}, acmyk = {alpha = 65535, cyan = 0, magenta = 0, yellow = 0, black = 0}, ahsl = {alpha = 65535, hue = 0, saturation = 0, lightness = 0, pad = 0}, array = {65535, 0, 0, 0, 0}}}, <RS_Flags> = {_vptr.RS_Flags = 0xe89228 <vtable for RS_Color+16>, flags = 0}, <No data fields>}}, updateEnabled = true, varList = {_M_t = {_M_impl = {<std::allocator<std::_Rb_tree_node<std::pair<QString const, QString> > >> = {<__gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<QString const, QString> > >> = {<No data fields>}, <No data fields>}, _M_key_compare = {<std::binary_function<QString, QString, bool>> = {<No data fields>}, <No data fields>}, _M_header = {_M_color = std::_S_red, _M_parent = 0x0, _M_left = 0x1cdd180, _M_right = 0x1cdd180}, _M_node_count = 0}}}}, entities = {<QListSpecialMethods<RS_Entity*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x1cf9880}, d = 0x1cf9880}}, subContainer = 0x0, static autoUpdateBorders = true, entIdx = 1, autoDelete = true}
        __for_begin = {i = 0x1cf9898}
        __for_end = {i = 0x1cf98a8}
#7  0x00000000005be012 in RS_Modification::moveRef(RS_MoveRefData&) (this=0x7fffffffc680, data=...) at lib/modification/rs_modification.cpp:3332
        addList = {<std::_Vector_base<RS_Entity*, std::allocator<RS_Entity*> >> = {_M_impl = {<std::allocator<RS_Entity*>> = {<__gnu_cxx::new_allocator<RS_Entity*>> = {<No data fields>}, <No data fields>}, _M_start = 0x1c6a7c0, _M_finish = 0x1c6a7c8, _M_end_of_storage = 0x1c6a7c8}}, <No data fields>}
#8  0x00000000005f087f in RS_ActionDefault::mousePressEvent(QMouseEvent*) (this=0x1af5f60, e=0x7fffffffcd30) at actions/rs_actiondefault.cpp:246
        m = {container = 0x1cdd0c0, graphic = 0x1cdd0c0, document = 0x1cdd0c0, graphicView = 0x1b10f20, handleUndo = true}
        data = {ref = {x = 102.75, y = 82.75, z = 0, valid = true}, offset = {x = 11.500000000000043, y = -20.749999999999979, z = 0, valid = true}}
#9  0x0000000000583776 in RS_EventHandler::mousePressEvent(QMouseEvent*) (this=0x1cde670, e=0x7fffffffcd30) at lib/gui/rs_eventhandler.cpp:101
#10 0x00000000006a3d8f in QG_GraphicView::mousePressEvent(QMouseEvent*) (this=0x1b10f20, event=0x7fffffffcd30) at ui/qg_graphicview.cpp:263
#11 0x00007ffff722de1f in QWidget::event(QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#12 0x00000000006a46a0 in QG_GraphicView::event(QEvent*) (this=0x1b10f20, event=0x7fffffffcd30) at ui/qg_graphicview.cpp:376
#13 0x00007ffff71e7acc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#14 0x00007ffff71ede27 in QApplication::notify(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#15 0x00007ffff64a19b0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#16 0x00007ffff71ec4f5 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib64/libQt5Widgets.so.5
#17 0x00007ffff7247d96 in  () at /usr/lib64/libQt5Widgets.so.5
#18 0x00007ffff724a6b3 in  () at /usr/lib64/libQt5Widgets.so.5
#19 0x00007ffff71e7acc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
[... 15 lines cut ...]
        show_splash = true
        windowX = 0
        windowY = 0
        maximize = false
        windowWidth = 1920
        files_loaded = false
        fileList = {<QList<QString>> = {<QListSpecialMethods<QString>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x1afa7b0}, d = 0x1afa7b0}}, <No data fields>}
        lang = {static null = {<No data fields>}, d = 0xff07d0}
        splash = 0xfb65b0
        langCmd = {static null = {<No data fields>}, d = 0xff0800}
        appWin = {<MainWindowX> = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff78ebc00 <QMainWindow::staticMetaObject>, stringdata = 0xae5d20 <qt_meta_stringdata_MainWindowX>, data = 0xae5e60 <qt_meta_data_MainWindowX>, static_metacall = 0x809d52 <MainWindowX::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}}, static staticMetaObject = {d = {superdata = 0xeb32a0 <MainWin [... cut ...]
        help0 = {static null = {<No data fields>}, d = 0xfae700}
        help1 = {static null = {<No data fields>}, d = 0xfae730}
        allowOptions = true
        argClean = {<QListSpecialMethods<int>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x7ffff6553080 <QListData::shared_null>}, d = 0x7ffff6553080 <QListData::shared_null>}}
        prgInfo = {d_ptr = {d = 0xfa6120}}
```

r-a-v-a-s, 2016-10-18:
It seems like the problem is in `void RS_Modification::deselectOriginals(bool remove)`.

If I change `document->addUndoable(e);`
to
```
document->startUndoCycle();
document->addUndoable(e);
document->endUndoCycle();
```
then it won't crash.

However if after moving I then

1. undo
2. move again
3. undo

it will crash, complaining about `if (e->isSelected()) {` a few lines above.

Hopefully someone else can review this... I don't really understand what the problem is.

kfoltman, 2016-10-18:
I was working on rewriting a lot of that area, but got stuck trying to fix undo on change properties of a block. If you give me more time, I can try to fix it.

r-a-v-a-s, 2016-10-31:
@kfoltman If you are stuck, then feel free to create an issue regarding your work;
or post at http://forum.librecad.org/LibreCAD-dev-f4361976.html