LibreCAD/LibreCAD #831: Elliptical arc angle ranges not handled correctly from dxf file

jsevy, 2016-10-18:
I've found that when drawing an elliptical arc defined in a dxf file, LibreCAD doesn't handle angle spans greater than 360 degrees (2 pi radians) correctly. For angles greater than 360 degrees, it drawa only the "overlap" - i.e., if asked to draw an arc from 0 to 450 degrees, it will draw just the arc from 0 to 90 degrees. Other DXF viewers don't seem to show this problem.

The problem is relevant because radians are used in dxf files for the angle ranges and rounding comes into the picture. If the angle span for an arc from 0 to 360 degrees is approximated as going from 0 to 6.2831855 radians, only a vanishingly small arc segment will be drawn since the rounding is slightly greater than 2 pi; on the other hand, an arc from 0 to 6.2831855 radians draws (effectively) the entire circle. The attached DXF file illustrates this.

[librecad_ellipse_arc_span_issue_illustration.txt](https://github.com/LibreCAD/LibreCAD/files/536501/librecad_ellipse_arc_span_issue_illustration.txt)

lordofbikes, 2016-10-18:
I've got your point, but I'm wondering if this is a theoretical question or if there is a practical use case for 450° arcs in 2D CAD.

LibreCAD uses the library libdxfrw for reading/writing DXF files. I don't remember that there were any issues with rounding and full ellipses in past. 
For a full ellipse libdxfrw uses the constant `6.283185307179586`. Rounding this to 7 digits, gives me `6.2831853` and not `6.2831855`. Thus I'm wondering, where this value come from. Maybe there is a problem with the DXF generator you use? Where does this DXF come from?

jsevy, 2016-10-18:
The result comes from a Java DXF generation library that I'm developing, which uses the standard Java Math library that's part of the Android development environment. I'm working with floats rather than doubles because that's the native type used in the Android graphics libraries.

The 6.2831855 comes from
    float endAngleRadians = (float) ( (startAngle + sweepAngle) * (Math.PI / 180));
with startAngle = 0 and endAngle = 360, and then using the standard Java string output operator to get the value as a string. Why it's represented as 6.2831855 instead of 6.2831853 I don't know...

I checked the behavior of LibreCAD to rounding at different points in the digit sequence 6.283185307179586; it's fine up until 6.2831853, even when the digits are rounded up, i.e., the full ellipse is drawn when using the value 6.28318531, but not when using 6.2831854 (though this admittedly isn't a correct rounding of the value anyway). It seems to get particular right around the float-double digit point: 6.28318531 works but 6.28318532 doesn't.

I can very likely work around this in my library by switching to double representations where needed; you can decide if it's worth looking into changing the LibreCAD (or libdxfrw) behavior to handle the general case where the angle extent exceeds 360. Just wanted to bring this to your attention since working with floats and standard libraries and rounding can seemingly bring on this behavior.

r-a-v-a-s, 2016-10-31:
> The 6.2831855 comes from
float endAngleRadians = (float) ( (startAngle + sweepAngle) * (Math.PI / 180));
with startAngle = 0 and endAngle = 360, and then using the standard Java string output operator to get the value as a string. Why it's represented as 6.2831855 instead of 6.2831853 I don't know...

Have you checked the quotient of `Math.PI / 180` for accuracy?

jsevy, 2016-10-31:
Just checked from within Eclipse:
    float temp = (float) (Math.PI/180);
    float tempTimes = (float) (360 * (Math.PI/180));
gives
    temp = 0.017453292
    tempTimes = 6.2831855
I'm using the Android SDK. Again, not sure why this gives something as far off from the correct value of 6.2831853 as it does... using a standard PC calculator gives  0.017453292 * 360  =  6.28318512

This is no longer an issue for me, as I switched angle quantities from float to double in my library and ellipses are now drawn as expected. But just wanted to let the LibreCAD developers know that ellipses don't handle angle extents greater than 360 in the way that would be expected, and it's possible to encounter this in a non-artificial situation, i.e., through rounding with floats and standard libraries...

lordofbikes, 2016-11-01:
In conclusion, the cause was using float for calculation and DXF file generation. This is a matter of precision, which ends for a float after 6th or 7th decimal digit, independent of decimal point. 

I can't see a reason for reducing precision in LibreCAD's DXF import, as this will impact precision in general.

I'm still wondering if the 450° arc is a real word problem or if this was only to clarify your observations.
The ellipse definition in DXF reference only defines 0 / 2PI for start / end angle for full ellipses. There is no hint about arcs spanning over 360° and I still can't imagine a use case in 2D CAD. Thus I'm fine for now with the angle normalization which is done in libdxfrw and leading to the behavior you have observed.

Anyhow, many thanks for pointing out your observations.

closed 2016-11-01 by lordofbikes