LibreCAD/LibreCAD #833: Object very small after ofen

anhhna, 2016-10-26:
hi all, im using LibreCAD 2.1.3 to open my file
https://drive.google.com/file/d/0B8Q2ZOp7YgmHTTdBek02ZE8zVk0/view?usp=sharing
but it's very small
https://drive.google.com/file/d/0B8Q2ZOp7YgmHQXYyRk1QZ0o5M2M/view?usp=sharing
need to zoom many times to see it's clearly
https://drive.google.com/file/d/0B8Q2ZOp7YgmHU2lEMEdqbmc3cnc/view?usp=sharing
plz help me to solve this problem, thank all so much

lordofbikes, 2016-10-26:
This is caused by the viewport. I'm not sure if this is only a DWG import or a general problem.

When you use View->Auto Zoom, you can zoom in in one step. Saving the zoomed drawing in LibreCAD (DXF) should open it later in zoomed state.

anhhna, 2016-10-27:
thank lordofbikes for replying but View->Auto Zoom not work, my DWG file above can open well on QCad or another softwares such as AutoCAD, TNT Atlas..., im trying to fixing it on LibreCAD, when you have free time, plz help me figure it out. Thank you!

lordofbikes, 2016-11-07:
I've found the cause why auto zoom doesn't work on your drawing. There is a frame with a couple of entities in origin at X0,Y0. Thus auto zoom is working generally, but not effective with your drawing.

Concerning the ignored view port, I expect the cause in the DWG import library as this works with DXF files. DWG file support is still in an early stage of development and the author is unresponsive for a while now.
I plan to enter maintaining libdxfrw, but this will take a while.

lordofbikes, 2016-11-08:
@anhhna, please don't wonder, the activities about title and label were done by me. It seems that this was a bug which should be fixed now, after my report to github support.

anhhna, 2016-11-08:
thank you very much @lordofbikes