LibreCAD/LibreCAD #834: change attributes of hatch makes it invisible

mstroehle, 2016-10-31:
After creating a Hatch (also very simple shapes created with the rectangle tool) it is not possible to change its attributes. Exactly it is not possible to change the assigned layer. After changing the layer by the attributes tool - the hatch is not visible anymore. Saving - Closing and reopen does not make it visible either. 
Librecad 2.1.3 on Windows 10.

poundpup1, 2017-01-29:
try to bring the object to the front. select object, then go to edit, drawing order, move to top.

fa201, 2020-01-29:
I see a strange behaviour.
I have 2 layers : _Rectangle_ and _Hatch_.
I created a rectangle on _Rectangle_ layer and draw some hatches with _Rectangle_ being active. Changing the attribute to _Hatch_ layer with _Rectangle_ being active does not work (the layer of hatches is still _Rectangle_).
But activating _Hatch_ layer and changing the attribute to _Hatch_ layer works.

Version: 2.2.0-rc1-155-g10d502db
Compiler: GNU GCC 7.4.0
Compiled on: Jun 15 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.3 LTS

slackuser0xae34, 2020-01-31:
I've looked at this, but am not sure I fully understand what the author is saying. If, I understand rightly, then:

SHORT ANSWER

After creating the rectangle entity, in the layer list, change the active layer to the "Hatch" layer, then create the hatch entity.

MEDIUM ANSWER

The hatch entity (not the "Hatch" layer) is an entity unto itself, separate from the rectangle entity. Unfortunately, the properties dialog associated with the hatch entity ("Choose Hatch Attributes") does not include the option to change the layer. The properties dialog associated with the rectangle entity ("Polyline") allows the layer to be changed. This is how the code is designed. 

LONG ANSWER

1) Open a new (blank) drawing.

2) In the layer list, create a new layer "Rectangle". 
    a) Set the default layer attributes (Context Menu/Right Click: "Edit Layer Attributes"). 
    b) Note here, there may be confusion as to what is an attribute and what is a property, and how each are assigned/used.

3) In the layer list, create a new layer "Hatch". 
    a) Set the default layer attributes (Context Menu/Right Click: "Edit Layer Attributes").
    b) Use a very different color than that used for the "Rectangle" layer, to aid comprehension.

4) In the layer list, set the "Rectangle" layer as the active layer.

5) In the drawing area, draw a rectangle.

6) In the drawing area, select the rectangle.

7) In the drawing area, apply a hatch to the rectangle (click the hatch icon, or "hatch" on the command line).
    a) The "Choose Hatch Attributes" box should appear.
    b) Set the attributes and click "OK".

8) The drawing area should have a rectangle with a hatch apparently "inside" it.
    a) Both the rectangle and hatch are entities on the "Rectangle" layer.
    b) A possible item of user confusion: the hatch entity is a separate entity from the rectangle entity. The rectangle was used to determine the limits of the hatch, but is not associated with the hatch in any other way (AFAIK).
    c) Meaning, that although the hatch appears to be inside and part of the rectangle, it is actually separate.
        1) This can be confirmed by carefully selecting either the rectangle or the hatch (but not both) and moving it. One can be moved without the other.
        2) Of course, selecting both (windowing) will naturally move both together. This may present the appearance of the rectangle and hatch being "connected".

9) Modify the rectangle entity properties (NOT the "Rectangle" layer attributes).
    a) On the command line type: "prop" (sans quotes), or
    b) "Modify" icon -> "Properties" option, or
    c) "Tools" -> "Modify" -> "Properties"

10) Select the rectangle, carefully lest the hatch be selected. 
    a) A "Polyline" box pops up with the rectangle properties.
    b) Included is a property for "Layer:". This can be changed from "Rectangle" to "Hatch", which moves the rectangle (only) from the "Rectangle" layer to the "Hatch" layer. 
    c) The hatch still appears inside the rectangle, but remains on the "Rectangle" layer. 
    d) Click "OK".
    e) Note, "prop" is still active in the command line box, asking for the next entity to modify.

11) Now select the hatch, carefully lest the rectangle be selected. 
    a) A "Choose Hatch Attributes" box pops up with the hatch properties.
    b) NOT included is a property for the "Layer:" (which WAS included on the "Polyline" box for the rectangle entity).
    c) Thus, hatches are restricted to the layer they are created on.
    d) AFAIK, the only way to change the layer of a hatch, is to delete the original hatch, change the active layer to the desired layer for hatches and recreate the hatch inside the rectangle.

Therefore, if I understand the author of this issue, what is being asked for is a way to change the layer that a hatch is on. That a "Layer:" selection pull-down be added to the "Choose Hatch Attributes" box.

On the other hand, perhaps I don't understand the problem. The line "But activating _Hatch_ ..." is adding something I don't follow. My 2 cents worth.

FYI, the hatch does not need to be created on the same layer as the rectangle. In the steps above, if the user changes the active layer to "Hatch" after step 5 (which creates the rectangle entity on the "Rectangle" layer) and before step 6 (in which the the rectangle entity is selected prior to creating the hatch), then the hatch will be placed on the "Hatch" layer in step 7.

BTW, Thanks for LibreCAD.

smrnv, 2020-02-01:
Author said that the problem with changing the hatch attributes after it was created. If you do it - hatch disappear. Not possible to change its color or move it to another layer.

hernot, 2020-02-15:
I do have similar problem when moving part of drawing, than hatch item disappears, becomes invisible even though i explicitly selected it as object/item to be moved. Same version and system combination as smrnv. The only solution i have found is to recreate hatch object anew.

Should i create a new issue or is it related.

m8ram, 2020-08-02:
I can also reproduce this problem.
Fedora LibreCad Version: 2.1.0.
Using the Properties dialog to change the scale, angle or pattern of an existing hatch seems to delete the hatch entirely.

fa201, 2021-02-02:
The issue is still there on last build:
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

Yet, I can change the hatch color matching the appropriate layer with _Attributes_ and then selecting the layer.

dellus, 2021-02-03:
Concerning the behaviour in 2.2.0 rc2 my experience is after changing the layer of a hatch with "Attributes", the hatch goes to the other layer but is still somehow connected with the original one. It keeps the line colour, type and width of the original, and if you switch the original layer unvisible, the hatch on the new layer will disappear also. But if you repeat the layer assignment with "Attributes", this connection gets cut and the hatch will have the desired properties.
So it seems with the first execution the command does not work completely, you have to do it twice!
If you know this it's a good workaround.