LibreCAD/LibreCAD #839: Some polylines become invisible when zooming

jborme, 2016-11-08:
When opening files saved with librecad 2.0.x within librecad 2.1.x, the elements (polylines...) included in layers which were not visible at the time of saving, will not be rendered on the screen when zooming in. It seems that toggling all layers to visible, then saving the file, will solve the problem.

Observation of the diff between the initial (buggy) file and the new one provides ideas on which flags changed:
[diff.txt](https://github.com/LibreCAD/LibreCAD/files/578817/diff.txt)

How to reproduce the problem:
1. Open the included DXF file
2. Set all layers to visible
3. Zoom in the pattern, observe that at some point most features disappear from the display (and come back when zooming out).

[20160831_unit_cell.zip](https://github.com/LibreCAD/LibreCAD/files/578826/20160831_unit_cell.zip)

All files saved in DXF R12 format.
Problem seen on gentoo linux with librecad 2.1.1 (gcc 4.9.3) and librecad 2.1.3 (gcc 5.4.0); also seen with the Windows binary distribution

r-a-v-a-s, 2016-11-08:
That sounds like a bug recently fixed.
https://github.com/LibreCAD/LibreCAD/issues/784

jborme, 2016-11-08:
Ok sorry, I did have a look at recently closed bug, probably missed this one.

closed 2016-11-08 by jborme