LibreCAD/LibreCAD #844: Trailing zeroes are incorrectly removed from dimension lines with integer precision

Sivullinen, 2016-11-24:
### Expected behavior

Correct value is always shown in dimension lines.

### Observed behavior

Dimension lines with integer precision (i.e. in the 'application preferences' the decimal part is set to have no digits) lose trailing zeroes from the **integer part** leading to massive errors in the dimension value.

### Steps to reproduce or sample file

See the attached files (drawing Test.dxf and output files Test_integer_precision.pdf and Test_decimal_precision.pdf taken with application preferences set to zero vs. non-zero digits in the decimal part. 

### Operating System and LibreCAD version info

Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7

The bug seems to be new, i.e. the SW has worked correctly in some previous version(s).

[Test_dxf.zip](https://github.com/LibreCAD/LibreCAD/files/611855/Test_dxf.zip)
[Test_integer_precision.pdf](https://github.com/LibreCAD/LibreCAD/files/611842/Test_integer_precision.pdf)
[Test_decimal_precision.pdf](https://github.com/LibreCAD/LibreCAD/files/611843/Test_decimal_precision.pdf)

Sivullinen, 2016-11-24:
Well, in fact the setting is in 'current drawing preferences', not in 'application preferences'.

Sivullinen, 2016-11-24:
This error might be duplicate of, or at least related to, the the closed issue #725 if the correction isn't yet in the version 2.1.3.

mudrd8mz, 2019-01-06:
I am still experiencing this serious bug in 2.1.3.

lordofbikes, 2021-08-23:
This was fixed with d730254350bf478afd8630c80eac69c01429786d in master

closed 2021-08-23 by lordofbikes