LibreCAD/LibreCAD #846: [feature] Changing fonts of all texts

jirislaby, 2016-11-26:
When I open any dwg drawing, all texts are rendered using the "standard" font. But Czech glyphs like "ěščřžýáíé" are rendered as diamonds, because they are not in the "standard" font.

It would be useful to have a feature to switch all used fonts to some other. In this case I would love to switch all texts to use "unicode" font.

jirislaby, 2016-11-26:
Alternatively, it would be enough if I could set the default font somewhere in the settings.

jirislaby, 2017-02-07:
The raw method which worked for me: `sed -i s/some_font/unicode/ drawing.dxf`

app4soft, 2024-08-31:
> Alternatively, it would be enough if I could set the default font somewhere in the settings.
>
> The raw method which worked for me: sed -i s/some_font/unicode/ drawing.dxf

It should be implemented as an option in "Application Prefernces" dialog, maybe as "Fonts substitute" table.

Also, there is built-in _"Same Properties"_ plugin, but actually it can't change text objects fonts.

Taylancan, 2025-05-14:
Hello, 
I import using your software and try to use LibreDWG as a work around.
Other than that Text-issue, I can import DWGs from almost seamlessly with LibreDWG Converter and edit it in LibreCAD.

A small code for usage of LibreDWG:
> @echo on
> echo Please copy in the path or draw the file into this terminal:
> set /p DataPath=
> echo You choosed: %DataPath%
> cd /d %DataPath%\..
> start /B C:\kpl\libredwg-0.13.3.7603-win64\dwg2dxf.exe -y %DataPath% >> dwg2dxf_by_LibreDWG.txt
> Pause

Thank you all for this great job.

My Process for correcting the text sized manually
1. double click on Text to open MText Window

[image]

2. change Font to unicode
4. change height to any proper size (mostly there is only a couple sizes used in the drawings I have)
5. do the exact same steps again and again and again and again and again and again ... for 100 Text positions? 😅🔫

That could be easily solved by having a function to edit **all selected MText** Items like we have for **Attributes**.
As a hint that would also make life easier when you like to change the font of older files.

[image]

Thank you again!
I would happily translate the new function into Turkish and German!

sand1024, 2026-09-26:
it's possible to change font via Properties widget (multi-entities)

closed 2026-09-26 by sand1024