LibreCAD/LibreCAD #850: Unicode Font Problems in dxf

RedDruid, 2016-12-07:
### Expected behavior
All text presents in Unicode, but LibreCad couldn't open dxf file correctly. I experimented with QCAD and it opened file as well.

### Observed behavior
All text replaced with '?'

### Steps to reproduce or sample file
**File**

[Unit.zip](https://github.com/LibreCAD/LibreCAD/files/635760/Unit.zip)

**And result QCAD vs LibreCad**

[image]
[image]

### Operating System and LibreCAD version info
Win10 and Version: 2.1.3

koolkhel, 2017-01-17:
First what you can do is to replace standard.lff inside LibreCAD/resources/fonts with unicode.lff - that will give you russian letters no problem. In your drawing each label is of different text style so LibreCAD is confused with that and it substitutes all fonts with "standard" which doesn't have any cyrillic. Not sure what is correct way to solve this in respect to LibreCAD architecture.

Second problem - and major one - is that you will see text attributes along with your text as LibreCAD doesn't seem to support text formatting attributes for now. I'll continue with investigations, but for now it seems as major amount of rework ahead.

AndreySV, 2017-05-04:
"Standard" is default font and is used as fallback for missing fonts. Why does it support only latin, numbers and greek characters? It'd be really nice to have all utf-8 characters supported in the default font. Then user at least could read text.

Symbian9, 2017-10-17:
Related issue
* https://github.com/LibreCAD/LibreCAD/issues/846

Fat-Zer, 2019-12-25:
Why not to make the «standard» font a placeholder rather than a real font and add a preferences entry to set it?

It's very easy to implement and fix 80% of problems for non-latin-1 dxf's...