LibreCAD/LibreCAD #853: Feature Request:  Provide Mapping Angular Layout

DocInvent, 2016-12-29:
### Provide for a 0-360 degree angles Clockwise beginning with 0 degrees straight up. This would be useful for platt and tract parcel and other property layouts.  Also, allow flipping this to follow the current Counter Clockwise with a beginning of 0 degrees straight up. 

### Only allowed to begin with 0 degrees directly right with angles Counter Clockwise

### New feature request.

### Ubuntu 16.04 with 
LibreCAD
Version: master
SCM Revision: 2.0.9
Compiler: GNU GCC 5.3.1
Qt Version: 5.5.1
Compiled on: Mar 24 2016

ngnmnt, 2017-01-15:
As above, this would be highly beneficial. 0 degrees UP, clockwise angles. Similar to a compass.
Hopefully easy to implement. Offset the base angle 90 degrees and change the +/- sign.

Essential to people working with charts, maps, GIS systems. This is the only reason I don't use LibreCAD so much. It would be great to see this as a new feature.

sand1024, 2026-09-26:
both clockwise and counterclockwise angles directions are supported, as well as user-define angles base direction

closed 2026-09-26 by sand1024