LibreCAD/LibreCAD #858: LibreCAD AppImage

probonopd, 2017-01-15:
LibreCAD today provides downloads for Windows and macOS at https://sourceforge.net/projects/librecad/files/ and there are PPAs for stable and daily versions for Ubuntu users. I suggest that in addition to that, an [AppImage](http://appimage.org/) should be provided, which would run on most Linux distributions and which could be automatically generated using the already existing binaries from the PPA.

Providing an AppImage would have, among others, these advantages:
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Just one format for all major distributions
- Works out of the box, no installation of runtimes needed
- Optional(!) desktop integration with `appimaged`
- Binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can GPG2-sign your AppImages (inside the file)

[Here is an overview](https://github.com/probonopd/AppImageKit/wiki/AppImages) of projects that are already distributing upstream-provided, official AppImages.

Do you think this would be useful?
I'd be happy to help producing an official LibreCAD AppImage.

Here are experimental ones for 2014-ish and later distributions:
https://bintray.com/probono/AppImages/LibreCAD#files

They were generated using these files:
* https://github.com/probonopd/AppImages/blob/master/recipes/meta/LibreCAD.yml
* https://github.com/probonopd/AppImages/blob/master/recipes/meta/LibreCAD-daily.yml

If you would like to generate an AppImage yourself, you can run on a debian/Ubuntu system:
```
wget -c "https://github.com/probonopd/AppImages/raw/master/recipes/meta/Recipe"
bash Recipe LibreCAD
# or
bash Recipe LibreCAD-nightly
```

Symbian9, 2017-10-17:
@r-a-v-a-s, providing autobuilt AppImage's as `continuous` build...
* https://github.com/probonopd/uploadtool

... could solve problem with non-existed latest versions of LibreCAD (at this moment: LibreCAD 2.1.3) on PPA
* https://github.com/LibreCAD/LibreCAD/issues/894

Also, its much easier setup `.travis.yml` for create AppImage with bundled bacported libs for Xenial 16.04 LTS, Trusty 14.04 LTS, etc.

P.S.: No need harms users built LibreCAD from source...
* https://github.com/LibreCAD/LibreCAD/issues/894#issuecomment-304482311

... if there is way how automate this process for current repo on Github using such amazing build tools as Travis CI, AppVeyour, etc. Users should draw in tool, not compile tool for drawing ;)

Symbian9, 2017-10-17:
> I'd be happy to help producing an official LibreCAD AppImage.

@probonopd, please, help setup official LibreCAD repo ...
* https://github.com/LibreCAD/LibreCAD/blob/master/.travis.yml

... for producing AppImages for `v.2.1.3` (as `latest release`, aka stable builds) and `master` (as `continous`, aka daily builds, aka `pre-release`)
* https://github.com/LibreCAD/LibreCAD/releases/latest
  * https://github.com/LibreCAD/LibreCAD/releases/tag/v2.1.3
* https://github.com/Symbian9/azpainter/releases/tag/continuous
  * https://github.com/Symbian9/azpainter/tree/master

REFERENCE:
* https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

probonopd, 2017-10-17:
Who in the project will take over maintainership?

antismap, 2017-11-18:
Hello, I'm interested in an AppImage because I run fedora. Do you need help ? I have background in ci (mainly in jenkins)

probonopd, 2017-11-18:
I'm also happy to send a Pull Request if at least one of the main developers like @dxli @Rallaz @r-a-v-a-s @lordofbikes is interested in it.

antismap, 2017-11-18:
This AppImage thing is awesome. I just run your script in an ubuntu VM with only build-essential (gcc was needed) and it all worked.

antismap, 2017-11-18:
sadly the ubuntu generated appimage segfaults under fedora 
```

(gdb) r
Starting program: /home/michael/Downloads/vm_shared/LibreCAD-2.0.2+nolibs.glibc2.14-x86_64.AppImage 
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.26-16.fc27.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Detaching after fork from child process 19536.
process 19532 is executing new program: /usr/bin/bash
Missing separate debuginfos, use: dnf debuginfo-install fuse-libs-2.9.7-9.fc27.x86_64 zlib-1.2.11-4.fc27.x86_64
Detaching after fork from child process 19541.
Detaching after fork from child process 19544.
process 19532 is executing new program: /usr/bin/bash
Missing separate debuginfos, use: dnf debuginfo-install bash-4.4.12-12.fc27.x86_64
Detaching after fork from child process 19549.
Detaching after fork from child process 19550.
Detaching after fork from child process 19551.
Detaching after fork from child process 19554.
Detaching after fork from child process 19555.
Detaching after fork from child process 19558.
Detaching after fork from child process 19561.
Detaching after fork from child process 19564.
Detaching after fork from child process 19565.
Detaching after fork from child process 19566.
Detaching after fork from child process 19567.
Detaching after fork from child process 19568.
Detaching after fork from child process 19569.
Detaching after fork from child process 19570.
Detaching after fork from child process 19571.
process 19532 is executing new program: /tmp/.mount_LibreCbfCCA7/usr/bin/librecad
Missing separate debuginfos, use: dnf debuginfo-install bash-4.4.12-12.fc27.x86_64
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/bin/librecad
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/82/48e4f45d39a219b0c24dd6fd9ec8caa0415087.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtHelp.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/83/9ab32c7cde74da092afd9096c8bf78eb8be489.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtSvg.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/f4/f663bd009e03087079ed0335a5d4c57c856feb.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtGui.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/f5/a98e2d91505fc03cde3142a8b37490bc2c8c11.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtCore.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/ea/c32a85c7607fde53016c145f6b3b61076660e6.debug
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libmuparser.so.2
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/43/19c18b93f774a0c6e1a1f40abb8a2fb3f045e0.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtSql.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/25/66a9caff1830b490aaf2eef752711ac42ddc25.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtNetwork.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/bc/12ab4497045eeaef580a4a6f25933ae153c328.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtCLucene.so.4
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/b5/dca0f0a3fa8755d65c301b8b9b9a7a93b7b539.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/lib/x86_64-linux-gnu/libpng12.so.0
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/33/6af8d05f01523cd07f0e50efb68f702e32886f.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libfreetype.so.6
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/86/a414c0d0e9a16656ec1c63292dd42660851602.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libXi.so.6
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/76/728e3cbc18394f4e5e93d150277f1f740bffe7.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libXrender.so.1
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/66/9c2699355514ca20ac340e2c5b6dde0366b4e1.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libXext.so.6
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/87/67b3fa70dcba92facf38b397755824a0889000.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libXt.so.6
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/64/4b6a8d4865c3459231b05e77e5aa5e6fb9ef98.debug
Missing separate debuginfo for /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libXau.so.6
Try: dnf --enablerepo='*debug*' install /usr/lib/debug/.build-id/47/4740350e3636b90972afc5c701d256de3d0c7d.debug
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
[... 13 lines cut ...]
#4  0x00007ffff7dea2e9 in _dl_open () from /lib64/ld-linux-x86-64.so.2
#5  0x00007ffff5208f96 in dlopen_doit () from /lib64/libdl.so.2
#6  0x00007ffff556a4df in _dl_catch_error () from /lib64/libc.so.6
#7  0x00007ffff5209715 in _dlerror_run () from /lib64/libdl.so.2
[... cut ...]
#19 0x00007ffff6c0a955 in QApplicationPrivate::initialize() () from /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtGui.so.4
#20 0x00007ffff6c0aa80 in QApplicationPrivate::construct(_XDisplay*, unsigned long, unsigned long) ()
   from /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtGui.so.4
#21 0x00007ffff6c0acef in QApplication::QApplication(int&, char**, int) () from /tmp/.mount_LibreCbfCCA7/usr/lib/x86_64-linux-gnu/libQtGui.so.4
#22 0x0000000000424376 in ?? ()
#23 0x00007ffff542d03a in __libc_start_main () from /lib64/libc.so.6
#24 0x0000000000428b07 in ?? ()
(gdb) 
```

probonopd, 2017-11-18:
Can you find out using `LD_DEBUG=libs` which libraries are loaded right before the crash happens?

antismap, 2017-11-18:
```
... 
    20109:	
     20109:	find library=libutil.so.1 [0]; searching
     20109:	 search path=/tmp/.mount_LibreCclivxP/usr/lib:/tmp/.mount_LibreCclivxP/usr/lib/x86_64-linux-gnu:/tmp/.mount_LibreCclivxP/lib:/tmp/.mount_LibreCclivxP/lib/x86_64-linux-gnu:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:		(LD_LIBRARY_PATH)
     20109:	  trying file=/tmp/.mount_LibreCclivxP/usr/lib/libutil.so.1
     20109:	  trying file=/tmp/.mount_LibreCclivxP/usr/lib/x86_64-linux-gnu/libutil.so.1
     20109:	  trying file=/tmp/.mount_LibreCclivxP/lib/libutil.so.1
     20109:	  trying file=/tmp/.mount_LibreCclivxP/lib/x86_64-linux-gnu/libutil.so.1
     20109:	  trying file=tls/haswell/x86_64/libutil.so.1
     20109:	  trying file=tls/haswell/libutil.so.1
     20109:	  trying file=tls/x86_64/libutil.so.1
     20109:	  trying file=tls/libutil.so.1
     20109:	  trying file=haswell/x86_64/libutil.so.1
     20109:	  trying file=haswell/libutil.so.1
     20109:	  trying file=x86_64/libutil.so.1
     20109:	  trying file=libutil.so.1
     20109:	 search cache=/etc/ld.so.cache
     20109:	  trying file=/lib64/libutil.so.1
     20109:	
     20109:	
     20109:	calling init: /lib64/libutil.so.1
     20109:	
     20109:	
     20109:	calling init: /lib64/libavahi-common.so.3
     20109:	
     20109:	
     20109:	calling init: /lib64/libavahi-client.so.3
     20109:	
     20109:	
     20109:	calling init: /lib64/libavahi-glib.so.1
     20109:	
     20109:	
     20109:	calling init: /lib64/libcrypto.so.10
     20109:	
     20109:	
     20109:	calling init: /lib64/libssl.so.10
     20109:	
     20109:	
     20109:	calling init: /lib64/libxml2.so.2
     20109:	
     20109:	
     20109:	calling init: /lib64/libgthread-2.0.so.0
     20109:	
Segmentation fault (core dumped)

```

probonopd, 2017-11-18:
Possibly we need to bundle glib, libgthread and friends. Please recreate the AppImage to see if https://github.com/AppImage/AppImages/commit/f779e10371d68a46fb34542ad4ff48a2edc08237 helps fix this.

antismap, 2017-11-18:
It works ! Thank you :)

lordofbikes, 2017-11-19:
@probonopd you're welcome to create a pull request if you want to take the challenge.
When the AppImage is built, we can handle the transfer to our sourceforge files section.

probonopd, 2017-11-19:
@lordofbikes would your consider switching the Travis build to a recent Qt version like this one:
https://launchpad.net/~beineri/+archive/ubuntu/opt-qt592-trusty

This would make the job _much_ easier for me. Basically we could follow https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci almost 1:1 then.

Symbian9, 2017-11-19:
> When the AppImage is built, we can handle the **transfer to our SourceForge files section**.

Also, it would be good to provide AppImage as `continous` release on Github too
* http://github.com/probonopd/uploadtool

lordofbikes, 2017-11-19:
I'm not sure what this means for our current CI configuration. 
When this will be a separate setup, I'm fine with your recommendations.
But If it will impact our current CI services, I'd prefer to use our own server to build the AppImage.

probonopd, 2017-11-19:
What @Symbian9 is trying to say is that you are currently using
https://github.com/LibreCAD/LibreCAD/blob/master/.travis.yml
to build on Travis CI.

Those builds could relatively easily be packaged up and posted on
https://github.com/LibreCAD/LibreCAD/releases
in a fully automated way.

Example:
https://github.com/Subsurface-divelog/subsurface/releases

KurtPfeifle, 2018-02-21:
@lordofbikes @probonopd:

Ping...

Are there any news on this issue?  Any chance for LibreCAD AppImages (for official releases as well as for continuous or "nightly" builds) any time soon?

probonopd, 2018-02-21:
@lordofbikes could you bump the Travis build to a recent Qt version likehttps://launchpad.net/~beineri/+archive/ubuntu/opt-qt592-trusty?

hubertbanas, 2020-11-22:
Is there any news on providing LibreCAD AppImages?

probonopd, 2021-04-27:
Hello everyone,it seems like there is renewed interest in having a LibreCAD AppImage. I'm happy to help.

It seems like LibreCAD does not have a `make install` target.
While this is a bit unusal, it just means that we may have to arrange the AppDir by hand by copying in the build products, in this structure:

```
./appdir/usr/bin/librecad
./appdir/usr/lib/...
./appdir/usr/share/librecad/...
```

Then it should be possible to run tools like [appimage-builder](appimage-builder.readthedocs.io), [go-appimage](https://github.com/probonopd/go-appimage/tree/master/src/appimagetool) `appimagetool -s deploy`, or `linuxdeployqt` on it.

lordofbikes, 2021-04-27:
Many thanks @probonopd!
With the files in the right place it basically workes for me with `linuxdeployqt`.
The plan for Travis is to switch back our config to Xenial and use Qt5.12 to build LibreCAD.
This is also my local setup when I successfully built the appimage.
With the appimage I have currently issues to use the bundled fonts and libraries, but I have to break now, will investigate this later.

probonopd, 2021-04-27:
You may be delighted to hear that with xenial being eol end of the month, the latest continuous build of `linuxdeployqt` should now have bionic as its new minimum version.

probonopd, 2021-04-27:
Also, you will need to fix the desktop file: `MimeType=image/vnd.dxf;` with a semicolon at the end.

lordofbikes, 2021-04-27:
Bionic is fine, this is our current setup for static analysis.
And yes, `linuxdeployqt` already told me about the `MimeType` issue.
I think I'll push my progress to a new branch soon, then everybody can have a look at it and help with testing.

probonopd, 2021-04-27:
Let's also see how https://github.com/probonopd/LibreCAD/actions goes.
That one uses [go-appimage](https://github.com/probonopd/go-appimage) instead of `linuxdeployqt`, and bundles _everything_ which should allow it to run on systems even older than the build system.
Let's compare results and take the one that works better.

Symbian9, 2021-05-21:
> Update: https://github.com/probonopd/LibreCAD/releases/download/continuous/LibreCAD-0.0-x86_64.AppImage got built 

@probonopd, Just tried and it works, BUT it seems like something wrong:

- On LibreCAD launch *in startup dialog* there are empty dropdown fields for "GUI Language" and "Command Language" (required translations files just NOT bundled inside `LibreCAD-*.AppImage`);
- In text input window I can't select font (highly likely LibreCAD just can't find path to its bundled fonts, i.e. `./squashfs-root/usr/share/librecad/fonts/*.lff`);
- On scheduled file autosave operation it shows warning message window *"Can't autosave the file"* (is there some issue with default autosave path access?).

lordofbikes, 2021-05-21:
@Symbian9 , all three points are not immediately related with AppImage creation.
I've created a branch https://github.com/LibreCAD/LibreCAD/tree/appimage with changes to our Travis CI build to create the AppImage, which works too, but with same issues.
Thus creating a LibreCAD AppImage works, but the LibreCAD executable could not handle the AppImage environment yet.
I'm working on a solution, we have to change the search locations for bundled content like plugins, fonts, libraries and translations.

When the executable finally works from AppImage, we can add the upload section to the Travis CI config.

lordofbikes, 2021-11-22:
@probonopd I just migrated our Travis CI build to GitHub Actions.
Then I migrated your latest action for LibreCAD appimage into this action.
The result is a `SIGSEGV` during AppImage creation and the resulting AppImage fails to start because of missing `libQt5Svg.so.5`.

I'm testing in my fork, you find the latest action logs at https://github.com/lordofbikes/LibreCAD/actions/runs/1492119889
The `SIGSEGV` occur in line 1778 of Create AppImage section.
The AppImage is available in Artifacts there.

Would you please be so kind and have a look at it.

probonopd, 2021-11-22:
```
2021/11/22 21:19:42 qt_prfxpath: /home/qt/work/install
2021/11/22 21:19:42 Got qt_prfxpath but it does not contain 'plugins'
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x801daa]
```

`qt_prfxpath` contains something go-appimage cannot deal with yet.

Where does the Qt come from?

lordofbikes, 2021-11-23:
Many thanks @probonopd!
In opposite to Travis CI build, I use jurplel/install-qt-action@v2 for Qt.
Presumably I have to source the Qt environment or install to different location.
I think I can figure this out, many thanks again.

lordofbikes, 2021-11-30:
Good things come to those who wait:
https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc3

Many thanks @probonopd for your support.

There are some minor improvements to do, concerning the build process, so I keep this open for now.
The AppImage itself should be safe to use.

eduncan911, 2021-12-03:
> Good things come to those who wait: https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc3
> 
> Many thanks @probonopd for your support.
> 
> There are some minor improvements to do, concerning the build process, so I keep this open for now. The AppImage itself should be safe to use.

works perfectly, thank you!

Symbian9, 2021-12-03:
Great news!

But there is still issue that `LibreCAD-*.AppImage` does not support system themes (and has no built-in dark theme)

Guess, that is common issue of AppImage so far, correct me @probonopd.

lordofbikes, 2021-12-03:
AppImage is a binary container format, which allows to run an application in different environments.
It does not affect the functionality of the application itself, at least not like you expect.
So these are LibreCAD issues, not related with AppImage and iirc we have already issues opened for them.

closed 2022-06-04 by lordofbikes