LibreCAD/LibreCAD #859: Zooming becomes slow when drawing polylines

jborme, 2017-01-15:
1. create new empty file, or open an existing file
2. Use tool Tools/Polyline/Polyline
3. Use the Zoom in our out buttons from the tool bar or using the mouse wheel.
Notice that zooming is visibly 3 times slower than usual speed of zooming (it becomes painful to use). Getting out of the polyline mode immediately stops this behaviour.

I noticed the following:
* It happens since 2.1 and is present in Master.
* It does NOT happen with the pre-built Windows version. It happens on my gentoo linux box. I cannot tell for other linux distros.
* It happens with SOME tools requiring the user to select some points, but not all tools. It happens with Polyline creation, Create Equidistant Polylines, Bevel, and Fillet. However It does happen with any of the the Circle and Ellipse drawing tools. It does not happen with Trim, Divide, Stretch modification tools.
* Zooming becomes slow even if document is empty (not connected to trying to zoom on complex layouts).
* Difficult snapping modes (such as Intersection) make it even slower, but even without snapping activated (and without grid displayed) it is still slower than what it should. Also, on the tools that do not have the problem (such as drawing circles), snapping is very fast even in intersection mode.

Because it does not happen with the pre-built version for windows, I would well accept it is connected with Qt5 under linux (not librecad code), but since it only occurs with some librecad functions, it could be that the point selection is coded a different way in different librecad funcrions so that there is a workaround. I tried Qt 5.6.2 and 5.7.1 with same results. I did not manage to compile a recent librecad with Qt4, it fails in lib/engine/rs_system.cpp:38:26 on class QStandardPaths.

Let me know if there is a way for me to produce useful debug output for this problem.

mpiffault, 2019-03-11:
I can confirm this with version :
```
Version: 2.2.0-rc1-105-gc2635841
Compiler: GNU GCC 7.3.0
Compiled on: Mar  9 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.2 LTS
```

It happens also with any dimensions tool, except the Leader one.

For Polyline tool, I narrowed the problem to the
`delete toolWidget;` line
at
`void QG_DialogFactory::requestPolylineOptions(RS_ActionInterface* action, bool on, bool update)`.
It takes around 450 ms on my machine.

The line has the exact similar instruction in
`void QG_DialogFactory::requestLineOptions(RS_ActionInterface* action, bool on)`
, but it takes a lot less time (0ms actually).

I'll try to dig further.

mpiffault, 2019-03-11:
Here is what I got for now, I focused on Polyline tool, but it seems the problem is similar for other tools :

Every now and then, `QG_DialogFactory::requestOptions` is called, which if Polyline tool is the current action, calls `QG_DialogFactory::requestPolylineOptions`.

Here, `QG_PolylineOptions` handled by the static field `QG_PolylineOptions* toolWidget` is deconstructed with `delete toolWdget`.
A new `QG_PolylineOptions` is instantiated into `toolWidget`
(`QG_PolylineOptions` is a QWidget displayed on top bar with options for the current polyline action).
 The deconstructor of `QG_PolylineOptions` is overloaded as :

```
QG_PolylineOptions::~QG_PolylineOptions()
{
    destroy();
}
```
and here is the `destroy()` function :
```
void QG_PolylineOptions::destroy() {
    RS_SETTINGS->beginGroup("/Draw");
    RS_SETTINGS->writeEntry("/PolylineMode", ui->cbMode->currentIndex());
    RS_SETTINGS->writeEntry("/PolylineRadius", ui->leRadius->text());
    RS_SETTINGS->writeEntry("/PolylineAngle", ui->leAngle->text());
    RS_SETTINGS->writeEntry("/PolylineReversed", (int)ui->rbNeg->isChecked());
    RS_SETTINGS->endGroup();
}
```

The main time consuming operations are here : each `RS_SETTINGS->writeEntry` call eats up to 100 ms.
What bugs me is that I can't narrow down to the inner calls which takes time. To me, it looks like a race condition somewhere, but for now it outreaches my analysis capacity.

Any help would be appreciated.

lordofbikes, 2019-03-12:
@mpiffault that's great, that you get to the heart of this issue.

There is no need to go deeper into this. ```RS_SETTINGS``` is the wrapper for ```QSettings``` class.
What happens here is to save current values of the tool option widget in a permanent store.
On Windows this takes place in the registry, but on Linux it is stored in ~/.config/LibreCAD/LibreCAD.conf.
This could explain the performance difference between Windows and Linux.
Each ```writeEntry()``` call cause a file operation on linux, searching and replacing the key/value pair and this is time consuming.

A cache mechanism for ```RS_SETTINGS``` could be a solution, but I think that would be the wrong attempt.

I wonder why the tool option widget is destroyed and recreated on each zoom iteration. This seems to be an issue in action handling. The tool option widget should be created on tool activation and destroyed on tool deactivation. Then this half second is less painful.
I also wonder if this happened on the way from 2.0 to 2.1 or if this is a property of QCAD.

So I would first check if this issue already exists in latest 2.0.x release.
When it is there too, there is no other chance than battling this in current version, understanding the action interface and program flow and search for mistakes in program logic.
If the issue is newer, ```git bisect``` can help to find the malicious commit and understand which change finally caused this issue.

xanderdin, 2019-03-12:
@lordofbikes 
> Each `writeEntry()` call cause a file operation on linux, searching and replacing the key/value pair and this is time consuming.

May be not as a fix for the issue, but as an improvement in general, what do you think about adding of something like
```C++
QVariant ret = readEntryCache(key);
if (ret.isValid() && ret == value) { return true; } 
```
to the very beginning of
https://github.com/LibreCAD/LibreCAD/blob/c2635841b751e1cea1b4cfc2129fea265b41241c/librecad/src/lib/engine/rs_settings.cpp#L96-L104

in order to skip writing at all if the key is already present and its value wasn't changed?

mpiffault, 2019-03-12:
@lordofbikes I made the same observation to myself about destruction / recreation of option widget on each mouse event. It's obvious for the zoom action, but it should not be the case either for each new point of a polyline or for each action of creating a dimension for example.

QSettings doesn't seam to be used for core tools of QCAD, and trying on my machine, it does not have the same behavior.

@xanderdin that's a good workaround in the meantime, thanks !

mpiffault, 2019-03-14:
Here is my proposition to fix this issue : #1079 

Two main corrections :
* Only destroy widget when explicitly needed (we already had the needed information at the right place).
* Do not destroy the widget when performing certain actions which immediately gives back hand to current action (e.g. most Zoom actions).

commit 46458dee (2019-03-17): Merge pull request #1078 from xanderdin/settings_write_skip

fix #859, fix #965, Skip re-saving already saved settings values
closed 2019-03-17 by lordofbikes