LibreCAD/LibreCAD #862: Compile error under LinuxCNC 2.7

effad, 2017-01-31:
### Expected behavior
Error free compile

### Observed behavior

```
> rli@linuxcnctest:~/LibreCAD$ git checkout tags/2.1.3
> Note: checking out 'tags/2.1.3'.
> ...
> rli@linuxcnctest:~/LibreCAD$ qmake librecad.pro 
> Project MESSAGE: We will be using CPP11 features
> rli@linuxcnctest:~/LibreCAD$ make
> ...
> g++ -c -pipe -g -std=c++11 -std=c++11 -O2 -Wall -W -fPIC -D_REENTRANT -DQT_WEBKIT -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I../../generated/lib/libdxfrw/moc -o ../../generated/lib/libdxfrw/obj/libdwgr.o src/libdwgr.cpp
> In file included from src/intern/dwgreader.h:20:0,
>                  from src/libdwgr.cpp:20:
> src/intern/dwgbuffer.h:45:13: error: looser throw specifier for ‘virtual dwgFileStream::~dwgFileStream()’
> src/intern/dwgbuffer.h:28:10: error:   overriding ‘virtual dwgBasicStream::~dwgBasicStream() noexcept (true)’
> src/intern/dwgbuffer.h:65:10: error: looser throw specifier for ‘virtual dwgCharStream::~dwgCharStream()’
> src/intern/dwgbuffer.h:28:10: error:   overriding ‘virtual dwgBasicStream::~dwgBasicStream() noexcept (true)’
> make[2]: *** [../../generated/lib/libdxfrw/obj/libdwgr.o] Fehler 1
> make[2]: Leaving directory `/home/rli/LibreCAD/libraries/libdxfrw'
> make[1]: *** [sub-libdxfrw-make_default] Fehler 2
> make[1]: Leaving directory `/home/rli/LibreCAD/libraries'
> make: *** [sub-libraries-make_default-ordered] Fehler 2
```

### Operating System and LibreCAD version info
```

> rli@linuxcnctest:~/LibreCAD$ cat /etc/debian_version 
> 7.9
> rli@linuxcnctest:~/LibreCAD$ uname -a
> Linux linuxcnctest 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
> 

```

lordofbikes, 2017-01-31:
What is the output of ```g++ --version``` and ```qmake --version``` on the LinuxCNC system?

effad, 2017-02-08:
```
rli@linuxcnctest:~$ g++ --version
g++ (Debian 4.7.2-5) 4.7.2
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rli@linuxcnctest:~$ qmake --version
QMake version 2.01a
Using Qt version 4.8.2 in /usr/lib/i386-linux-gnu
```

lordofbikes, 2017-02-08:
I remember issues #741/#742, thus Qt 4.8 should be fine for 2.1.3, but I'm not sure about g++ 4.7, have to do some more investigations

fa201, 2023-08-09:
@lordofbikes Is this issue corrected ?

sand1024, 2026-09-26:
too old

closed 2026-09-26 by sand1024