LibreCAD/LibreCAD #865: Appending a node to polyline duplicates the original

jirislaby, 2017-02-07:
### Steps to reproduce or sample file

1. draw a polyline (e.g. 3 nodes)
2. append a node to polyline

### Expected behavior

One polyline with the appended node.

### Observed behavior
Two polylines: the correct one and the original one

### Operating System and LibreCAD version info
openSUSE and librecad-2.1.99+git.20161127-68.1.x86_64 -- the git snapshot 0d239ca908c5cb07b8b05f67be2c5cd241ec8fc1.

Akranen, 2017-10-21:
I believe this one is not happening anymore.

jirislaby, 2018-03-11:
It still happens with librecad-2.1.99+git.20180202-74.6.x86_64.

fa201, 2020-04-14:
I created a polyline and a point to append.
When I click on the point I have the message "No Entity Found", even though the snap worked.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

jirislaby, 2020-04-15:
@fa201 How does it relate to the original bug?

fa201, 2020-04-15:
It is possible that the correction of the bug created another bug. So I reported here and if it is not the right place then another issue will be created.

dxli, 2023-08-16:
cannot reproduce this one.

closed 2023-08-16 by dxli