LibreCAD/LibreCAD #866: dwg imported rotation is pi/2

jirislaby, 2017-02-07:
I imported a drawing from DWG and all texts which were 90 degrees rotated are rotated only 1.570796326794897 degrees. This is exactly PI/2. So there is a missing conversion from RAD to DEG somewhere on the path.

### Operating System and LibreCAD version info
openSUSE and librecad git snapshot 0d239ca908c5cb07b8b05f67be2c5cd241ec8fc1.

webmite, 2017-05-20:
This may be related to pull #890.  
It may not be a missing conversion...it may be a missing translation. 
If the orientation of the text has not taken into account ANGBASE and ANGDIR this could
be a side-effect.

webmite, 2017-05-29:
Can you provide a simple sample DWG file. 
Say a simple Layer 0 Text element with the word 'hello' rotated 90 degrees.
That would be sufficient for testing purposes.

jirislaby, 2017-06-09:
I cannot generate one given I don't have access to acad. But this is the drawing, I have problems with:
http://www.fi.muni.cz/~xslaby/odhad/pp/projekty/zmeny_10_2016_projektantovo_nepouzitelne.dwg

vittorio88, 2018-06-20:
Hello,

I suspected you may be running into the same issue I am, so I ran your file through a debugger to verify. The issue is the same as mine, the underlying library libdxfrw cannot read that value `oblique` from each style in the dwg file.
I don't know why you are seeing this specific number 1.57... , but I believe it's because at the lower level it is reading garbage. This issue occurs specifically in the `DRW_Textstyle::parseDwg()` function where the style table is parsed. More specifically, it seems that one too many (or too little) bytes are read from the buffer of the styles table of the dwg file.

I am unsure if importing text styles is broken completely or if it occurs only on certain files. It would be useful to have a sample file where importing text styles does work.

You can see in the following image the values being read which are extraordinarily large or small.
[image]

I too am working on fixing this, but I don't really know where to go from here...