LibreCAD/LibreCAD #868: Infinite loop while manipulating construction lines (quadratic form)

damienflament, 2017-02-15:
### Expected behavior
Select the line.

### Observed behavior
 * Infinite loop (100% CPU usage),
 * unusable UI (need to kill the _librecad_ process.

### Steps to reproduce or sample file
 * Create a new file,
 * make a vertical line at position (0,0),
 * set the current layer as a _construction_ layer,
 * pan the view to the up-right to hide the previously constructed line,
   - **note**: after this step, a line appears starting at the upper-right corner of the viewport and ending
      near the lower-left corner.
 * try to select all objects (_Ctrl-A_).

**N.B:** With a previous LibreCAD version (maybe 6 month ago), panning the view containing construction lines always triggers the same issue. Now (with current stable or the Git version), it seems to be more unpredictable. The steps described above definitely "work".

### Operating System and LibreCAD version info
 * ArchLinux `_uname -a`: 
   > Linux _<hostname>_ 4.9.8-1-ARCH SMP PREEMPT Mon Feb 6 12:59:40 CET 2017 x86_64 GNU/Linux
 * LibreCAD:
   - version 2.2.0 alpha (from current ArchLinux User Repository package _librecad-git_),
   - compiled for x86_64 architecture.

### Debug log
Logs created using command `librecad --debug 6 &>librecad.txt`: [librecad.txt](https://github.com/LibreCAD/LibreCAD/files/776303/librecad.txt)

LibreCAD seems to loop on computing _quadratic forms_. But the logs are not populated anymore when it starts infinite looping.

r-a-v-a-s, 2017-02-16:
Confirmed... odd behavior,

For anyone interested in solving this,
I suggest looking at 
`RS_GraphicView::drawEntity(RS_Painter *painter, RS_Entity* e, double& patternOffset)`
and
`RS_Line::draw(RS_Painter* painter, RS_GraphicView* view, double& patternOffset)`
and
`RS_Selection::selectAll(bool select)`

jamesruan, 2017-09-07:
confirmed.
I lost half an hour's work of this odd line, go through the top-right to bottom-left.

fa201, 2023-08-09:
I cannot reproduce this with
Version: 2.2.0.2
Compiler: GNU GCC 9.4.0
Compiled on: Jul 29 2023
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Ubuntu 20.04.6 LTS

sand1024, 2026-09-26:
obsolete issue

closed 2026-09-26 by sand1024