LibreCAD/LibreCAD #873: Can not open DWG file

NguyenTrongThinh, 2017-03-05:
I use LibreCAD to draw some shape then save file to disk, then i restart LibreCAD and choose file -> open to open my old file but i see not thing on screen except point (0, 0). 
OS: windows 10 64 bit.
LibreCAD v2,1,3

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

r-a-v-a-s, 2017-03-05:
Your steps would produce a DXF (not DWG).

What happens if you use view -> auto-zoom (double-click the scroll wheel)?

NguyenTrongThinh, 2017-03-06:
Sorry for wrong file extension. It is "dxf". I try "view -> auto zoom" but i still can not see anything on screen. 
FYI, I have 2 PC, one ASUS and the other is HP, The problem happens on my HP computer. LibreCAD works fine when I open file on my ASUS PC. All of them use Windows 10 x64, LibreCAD 2.1.3.

r-a-v-a-s, 2017-03-06:
My guess would be it's a Qt / Windows 10 issue.
Did you download from sourceforge? Is it still using Qt 5.4? (help -> about)
Building with Qt 5.7 _might_ solve it, because a lot of Windows 10 compatibility was added
after Qt 5.4.

NguyenTrongThinh, 2017-03-06:
Thank for your response! i will try to build LibreCAD from source using Qt 5.7.0 (MingW 5.3) then debug on my HP PC, if the problem does not happen, I will report for you and close this issue.

NguyenTrongThinh, 2017-03-06:
I have built LibreCAD (branch Master) with Qt 5.7.0 (mingW32 5.3.0) and tested. The problem I report did not happen. So i think it is Qt issues, I closed this issue

closed 2017-03-06 by NguyenTrongThinh