LibreCAD/LibreCAD #875: Unable to import DXF files into AutoCAD

kspartin, 2017-03-31:
I have created a bunch of drawings in LibreCAD 2.1.3. My files are unable to be opened by AutoCAD 2012 or Autodesk DWG TrueView 2016.

### Expected behavior
Expect AutoCAD to import the DXF file.

### Observed behavior
The following error was encountered while reading
in LEADER starting at line 14610:
Leader entity with no vertices
Invalid or incomplete DXF input -- drawing discarded.
Press ENTER to continue:
Regenerating model.
**** System Variable Changed ****
1 of the monitored system variables has changed from the preferred value. Use SYSVARMONITOR command to view changes.
*Cancel*
*Cancel*
_RIBBON
*Cancel*

### Steps to reproduce or sample file
Attached file test.zip contains a DXF file that exhibits this behavior. Neither AutoCAD 2012 or Autodesk DWG TrueView 2016 can open the file.
[test.zip](https://github.com/LibreCAD/LibreCAD/files/886737/test.zip)

The file test_2.zip contains a DXF file that was converted by a program called "Any DWG DXF Converter". AutoCAD can open that file, but as you can see, there are issues with text and dimensions. In addtion, Autodesk DWG TrueView 2016 can open file test_2.dxf.
[test_2.zip](https://github.com/LibreCAD/LibreCAD/files/886742/test_2.zip)

### Operating System and LibreCAD version info
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows

MoQ93, 2018-07-06:
is there any update on this?

ddanila, 2019-07-31:
Hi the same issue with
Version: 2.1.3
Compiler: Clang 8.0.0
Compiled on: Sep 22 2016
Qt Version: 5.6.1
Boost Version: 1.60.0
System: OS X 10.13

Importing to AutoCAD LT 2011

hoxnox, 2021-12-07:
The same issue

Version: 2.1.3
Compiler: GNU GCC 10.3.0
Compiled on: Nov 13 2021
Qt Version: 5.15.2
Boost Version: 1.76.0

Importing to AutoCAD LT 2022

hoxnox, 2021-12-07:
The same for

Version: 2.2.0-rc3
Compiler: GNU GCC 9.3.0
Compiled on: Nov 28 2021
Qt Version: 5.12.9
Boost Version: 1.75.0

motla, 2022-03-18:
Same issue here. A workaround was to open the LibreCAD files with the software [CST CAD Navigator](https://fr.cadsofttools.com/products/cst-cad-navigator/), which opened them well, then save them from CST CAD.

lordofbikes, 2022-03-19:
The issue with the submitted `test.dxf` is, that it contains 2 `LEADER` without coordinates.
`test.dxf` has 28 `LEADER` and the re-saved `text_2.dxf` has 26, thus re-saving eliminated the malicious `LEADER` entities.
This succeeds for me even with latest LibreCAD version 2.2.0-rc3-20-ge829bce1.

So the question here is, how is it possible to create `LEADER` entities without coordinates.

I can't reproduce this and have no idea, how I should achieve a `LEADER` without coordinates.

Before anybody posts more _same issue_ or _me too_ messages, provide the steps to make this issue reproducible.
Don't provide old files which don't open in AutoCAD, the issue may be already solved.
Use your current LibreCAD, create a new drawing with `LEADER` and save it.
When it fails to open in AutoCAD, submit:
- your LibreCAD version
- the dxf file
- exact steps how you created it
- have you used undo/redo or any other modifying tools?

oslucchi, 2022-09-13:
In my case it seems to have to do with layers.
In the attached zip file you can find 2 dxf: withLayers and noLayers. I tried uploading the first to AutoCAD and it has been rejected.
I then removed all of the newly created layers changing the layer properties of all objects to the default 0
The file has then been save and, at that point, it could be opened by the AutoCAD viewer.
LibreCAD version 2.1.3
modifying tools were used originally (trim two; lenghten; change properties) but those should have affected also the noLayers version in case...
Could this be of any help?

[dxfFiles.zip](https://github.com/LibreCAD/LibreCAD/files/9558387/dxfFiles.zip)
[image]
[image]

lordofbikes, 2022-09-13:
@oslucchi I tried to open `withLayers.dxf` with Autodesk TrueView, which was more responsive about the issues.
The problem is just the layer name `12,5MM` which is rejected by AutoCAD. Renaming the layer to `12_5MM` solved this for me. `12.5MM` didn't work either. So AutoCAD seems to dislike `,` or `.` in layer names.

There was also another issue with an unknown code in dimension style table. I can't say where this comes from.
Possibly it is because of the DXF versions, `withLayers.dxf` is DXF R12 (1992!) and `noLayers.dxf` is DXF 2007 format. Possibly the unknown code was invalid for R12, but valid for 2007.

So open with latest 2.2.0-rc4 version, rename this one layer and save in DXF 2007 format might have been enough to make the file open in AutoCAD.