LibreCAD/LibreCAD #876: Can not open the file <name>.dwg  -  Please check its existence and permissions.

cyborg-x1, 2017-04-03:
### Expected behavior
Select a dwg file, see loaded dwg file contents, work on it

### Observed behavior
Select a dwg file, see loaded dwg file contents, message box says:

> Can not open the file file.dwg
> Please check its existence and permissions

file contents disappear after clicking OK in this message box.
(But in the background you could see the file was indeed loaded and contents
are deleted again)

### Steps to reproduce or sample file
see above

### Operating System and LibreCAD version info
Ubuntu 16.04
LibreCAD stable (librecad_2.0.9-2_amd64.deb) and daily (2.1.1+20170330.431-0~ubuntu16.04.1)

historicbruno, 2017-05-29:
I believe this rather unhelpful message is telling you the DWG import failed.  Try running LibreCAD with the --debug6 option, to see if any useful output is logged. I'm in the process of debugging with some of my DWG files that cause this error, but not all of them do.

juanda-097, 2017-07-14:
I have the same problem but --debug6 says this:
`File C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\ACcuagyr.jpg doesn't exist.
File /home/juanda097/C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\ACcuagyr.jpg doesn't exist.
File /home/juanda097/C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\ACcuagyr.jpg doesn't exist.
linking image: OK
RS_FilterDXFRW::linkImage
File C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\AACCUAGYR nuevo logo.JPG doesn't exist.
File /home/juanda097/C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\AACCUAGYR nuevo logo.JPG doesn't exist.
File /home/juanda097/C:\varios planos\Planos\ORGANIZACION DEL 2006\LOGOS\AACCUAGYR nuevo logo.JPG doesn't exist.
linking image: OK
RS_FilterDXFRW::linkImage
File H:\Municipio de Girardot\Actualizacion POT\Plano SIG POT\Logos\Escudo.png doesn't exist.
File /home/juanda097/H:\Municipio de Girardot\Actualizacion POT\Plano SIG POT\Logos\Escudo.png doesn't exist.
File /home/juanda097/H:\Municipio de Girardot\Actualizacion POT\Plano SIG POT\Logos\Escudo.png doesn't exist.
linking image: OK
RS_FilterDXFRW::fileImport: reading DWG file: OK
QG_DialogFactory::commandMessage
QG_DialogFactory::commandMessage: OK
QG_DialogFactory::commandMessage
QG_DialogFactory::commandMessage: OK
RS_FilterDXFRW::printDwgError: DRW::BAD_READ_TABLES
Cannot open DWG file '/home/juanda097/9-DIVISION PREDIAL_.dwg'.
RS_FilterDXFRW::~RS_FilterDXFRW(): OK
QC_MDIWindow::slotFileOpen: failed
QC_MDIWindow::slotFileOpen: OK
QXcbConnection: XCB error: 3 (BadWindow), sequence: 2996, resource id: 35652110, major code: 40 (TranslateCoords), minor code: 0
QC_ApplicationWindow::slotWindowActivated begin
`

SkellaMental, 2019-05-01:
Bump (are we allowed to bump threads?)
Any news/update on this?
Cheers

historicbruno, 2019-05-06:
The only thing that could be done in general, is to improve the error dialog to show the actual reason for the import failure, without having to resort to debug logging.

Otherwise, it depends on the specific DWG in question, which will likely need to be provided here for further debugging. If someone could create a minimal test DWG that reproduces such errors, it would surely help! :-)

lordofbikes, 2019-05-06:
The problem is that the read/write methods are ```bool``` and simply return ```true``` or ```false```.
So we have to revamp *libdxfrw* library to supply more expressive error codes here.
Also, the DWG import capabilities are still very rudimentary and require attention too.

AkuOkka19, 2019-11-13:
I have an 80 KB file that reproduces this error that I can share with you if it helps.

gerazov, 2020-11-27:
I'm experiencing the same problem. The file opens with [ODA Drawings Explorer](https://www.opendesign.com/guestfiles/oda_drawings_explorer). Here's the file: [sample.zip](https://github.com/LibreCAD/LibreCAD/files/5607403/sample.zip)

lordofbikes, 2021-01-31:
Follow up to #988

closed 2021-01-31 by lordofbikes