LibreCAD/LibreCAD #879: Corner rounding / fillet  trimming

Haecksler, 2017-04-17:
### Expected behavior  conner should be trimmed as in older versions 

### Observed behavior : by choosing first and second entity it makes a double line instead trimming the corner 

### Steps to reproduce or sample file

### Operating System and LibreCAD version info Linux Mint 18.1 ; 64 bit  Libre CAD 2.0.9-2

webmite, 2017-11-24:
using fillet this is the behaviour I observe....selecting the first entity  behaves fine...but selection of the 2nd affects which direction the arc will go to join the lines.  If the mouse is 'between' the lines forming the acute angle then the fillet will form correctly....If the mouse is 'outside' the lines forming the acute angle then the fillet will be malformed.

It should not matter which side of the entity you are on when you click the mouse...both cases should produce the first 'correct' action.  2nd behaviour should be blocked.
[image]
[image]
[image]

lordofbikes, 2017-11-24:
I'm not sure if there was an issue with fillet tool in the past and I can't find any fix in git log. 
Maybe it is related with the _Trim_ checkbox in the tool bar. The checkbox and the fillet Radius input appear on top of the drawing area, when the fillet tool is selected.
[image]
When Trim is unchecked, the tool works as reported. Maybe it was unchecked by accident?

For latest release I can confirm that it works fine, thus I'll close this issue.

@webmite your observations are correct, but this is how the tool was designed. I don't see the 2nd fillet as malformed, it may be a desired option to draw the arc this way. I think there is no reason to restrict the tool as the preview shows the drafter what happens next.

closed 2017-11-24 by lordofbikes