LibreCAD/LibreCAD #880: compilation termination -> fatal error: stdlib.h not found

RomankoMikhail, 2017-04-19:
### Expected behavior
Compiled binaries of LibreCAD from git repo sources.
### Observed behavior
Compilation termination. fatal error: stdlib.h not found
### Steps to reproduce or sample file
1) git clone https://github.com/LibreCAD/LibreCAD.git
2) cd LibreCAD
3) qmake
4) make -j4

### Operating System and LibreCAD version info
Linux Acer-pc 4.9.22-1-MANJARO #1 SMP PREEMPT Wed Apr 12 21:17:49 UTC 2017 x86_64 GNU/Linux

GIT sources from April 19 2017 18:00 UTC+3

GCC:
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/lto-wrapper
ARCH: x86_64-pc-linux-gnu
CONFIGURATION: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release
gcc version 6.3.1 20170306 (GCC)

QT: Using Qt version 5.8.0 in /usr/lib

r-a-v-a-s, 2017-04-19:
Probably because the merging of https://github.com/LibreCAD/LibreCAD/commit/ed63b8a1d77c4928674ddfff2a929916e1020037
didn't consider the reason why the line was added (https://github.com/LibreCAD/LibreCAD/commit/35a1a9e6fb8985993b3a8c3f1637d73331789794).

r-a-v-a-s, 2017-04-19:
You can also add "QMAKE_DEFAULT_INCDIRS=\\"
to the qmake arguments in Creator.

or (untested)

```
vi .qmake.cache
CONFIG += release
QMAKE_DEFAULT_INCDIRS +=
QMAKE_LRELEASE="/usr/bin/lrelease"
```

lordofbikes, 2017-04-28:
This issue arise with the use of GCC 6, building with GCC 5 still works fine. There has something changed with include search order in GCC 6, have to do more readings to see what we can do to solve this.

JeanLucCoulon, 2017-05-08:
Hi,

There is a good explanation of the problem here:
https://patchwork.ozlabs.org/patch/674113/

And a quick fix (I don't know the side effects) is to add the following line in ./librecad/src/src.pro
QMAKE_CFLAGS_ISYSTEM = ""

This avoid the -isytem flag to be passed which is enabled by default

Regards

Jean-Luc

lordofbikes, 2017-05-08:
Thanks again Jean-Luc!

I found no side effects with gcc 5 and 6 on different systems and thus applied your suggestion.

When the daily ppa on Launchpad will build again for Yakkety with this change I'll close this issue later.

closed 2017-05-09 by lordofbikes