LibreCAD/LibreCAD #883: Document Surveyor's Units in Current Drawing Preferences

elil, 2017-05-07:
### Expected behavior

Navigating through the interface Options-->Current Drawing Preferences-->unit(tab)-->Angle(section) and selecting "Surveyor's units" does not give an example or preview.  Like the other choices, an example is expected in the preview section.  

### Observed behavior

The preview is blank.  

### Steps to reproduce or sample file

Open program, then navigate to Options-->Current Drawing Preferences-->unit(tab)-->Angle(section) and selecting "Surveyor's units".  Watch the Angle preview while doing this and change it between a few things to see the presence and absence of the preview.  

### Operating System and LibreCAD version info

MacOS, 10.12.4
 
Version: 2.1.3
Compiler: Clang 8.0.0
Compiled on: Sep 22 2016
Qt Version: 5.6.1
Boost Version: 1.60.0
System: OS X 10.12

### Additional information

I looked [here](https://github.com/LibreCAD/LibreCAD/blob/3d4adab1081e683e34d7bd519f1d9fa5b8f5cd2b/librecad/src/lib/engine/rs.h#L511) and [here](https://github.com/LibreCAD/LibreCAD/blob/3d4adab1081e683e34d7bd519f1d9fa5b8f5cd2b/librecad/src/ui/forms/qg_dlgoptionsdrawing.cpp#L687) trying to figure it out and am still not quite sure.  

It might be quadrant bearing like N30d31'32"E (which is east or west bearing from north or south see also [#5 in wikipedia entry](https://en.wikipedia.org/wiki/Bearing_(navigation)#Bearing_measurement) or any search for 'convert azimuths and bearings' like [this](http://academic.brooklyn.cuny.edu/geology/leveson/core/linksa/comp.html)).  N30d31'32"E is pretty close to the number being used in the other examples, however, it is in the NE quadrant which makes it less obviously meaningful.  A better overall example might be in a different quadrant like (in Deg/min/sec) 149(0xB0) 28' 28" which in bearings would be S30d31'32"E.  

The other thing it could be (either instead of or in addition to a quadrant bearing) is that many surveying calculators and programs use a notation like 149.2828 which "obviously" (not obvious at all and very confusing) means 149(0xB0) 28' 28" and not 149.2828 or (in Deg/min/sec) 149(0xB0) 47'44"!

lordofbikes, 2017-05-12:
It seems that there is not only the preview missing. On the quick, the implementation of surveyors units looks incomplete at all. Have to investigate, when this was added and how much is missing.

lordofbikes, 2017-05-30:
With the fix #889 from @webmite this issue is closed.

Generally the support for [GIS](https://en.wikipedia.org/wiki/Geographic_information_system) stuff is still very poor in LibreCAD. There were some guys working on a fork for better GIS support in LibreCAD, but I'm not sure about its state.

closed 2017-05-30 by lordofbikes