LibreCAD/LibreCAD #891: LibreCAD freezes opening a dxf previously saved

luislfc, 2017-05-21:
### Expected behavior
It should open the file normally, show all lines and not freeze while we "navigate" the file.

### Observed behavior
LibreCAD freezes and the CPU jumps to 100%. It will never recover from this state

### Steps to reproduce or sample file
1- Open the sample dxf file,
2- LibreCAD will not show all lines but it still is not frozen
3- select "view -> auto zoom" and it will freeze

There are other ways to replicate, like zooming out with the mouse, but the easiest way to replicate is the described one.

### Operating System and LibreCAD version info
Linux version 4.8.0-1-amd64
CPU: Intel(R) Core(TM) i7-3630QM CPU @ 2.40GHz
Version: 2.1.2
Compiler: GNU GCC 6.3.0
Compiled on: Sep 17 2016
Qt Version: 5.7.1
Boost Version: 1.62.0
System: Debian GNU/Linux 9 (stretch)

### Notes
The dxf file was zipped as the issue tracker doesn't accept dxf files to be uploaded.

[Transistors.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/1017101/Transistors.dxf.zip)

metrologymaster, 2017-05-22:
I have experienced the same issue while using the Windows version running under both Win7 and Win10.

Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7

gratefulfrog, 2017-05-31:
I have the same problem on Ubuntu Linux 64it when zooming on a previously created dxf.

Version: 2.2.0-alpha
Compiler: GNU GCC 5.4.0
Compiled on: May 31 2017
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Ubuntu 16.04.2 LTS

dxli, 2023-05-22:
The file Transistor.dxf  contains a polyline bulge of positive inf bulge.

dxli, 2023-05-22:
The infinity(saved as -1.633123935319537e+16, line 2716 of the file Transistor.dxf)
 bulge is found for the polyline bulge field (code=42).

The definition of bulge is $\tan(\frac{\alpha}{4}$, where $\alpha$ is the angular size of the arc. Conceputally, zero angular size is equivalent to $2 \pi$, but a bulge size of $\pm\infinity$ cannot be used to construct an arc.

The fix is mostly for the symptom, but the root cause may still need to be fix to avoid infinity bulges when saving to dxf.

If the steps to create the problematic dxf is known, please reopen this issue.

closed 2023-05-22 by dxli