LibreCAD/LibreCAD #892: clang: error: unknown argument: '-fext-numeric-literals'

ryandesign, 2017-05-23:
### Expected behavior

Compiles successfully.

### Observed behavior

Fails to compile:

```
/usr/bin/clang++ -arch x86_64 -c -Os -stdlib=libc++ -arch x86_64 -g -std=c++11 -fext-numeric-literals -O2 -Wall -W -DDWGSUPPORT -DQC_APPDIR="LibreCAD" -DLC_VERSION="2.2.0-alpha" -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/opt/local/libexec/qt4/share/mkspecs/macx-g++ -I. -I../../generated/librecad/moc -I/opt/local/libexec/qt4/Library/Frameworks/QtCore.framework/Versions/4/Headers -I/opt/local/libexec/qt4/Library/Frameworks/QtGui.framework/Versions/4/Headers -I/opt/local/libexec/qt4/Library/Frameworks/QtSvg.framework/Versions/4/Headers -I/opt/local/include -I../../libraries/muparser/include -I../../libraries/libdxfrw/src -I../../libraries/jwwlib/src -Icmd -Ilib/actions -Ilib/creation -Ilib/debug -Ilib/engine -Ilib/fileio -Ilib/filters -Ilib/generators -Ilib/gui -Ilib/information -Ilib/math -Ilib/modification -Ilib/printing -Iactions -Imain -Itest -Iplugins -Iui -Iui/forms -Iui/generic -I../res -I/opt/local/libexec/qt4/Library/Frameworks/QtSvg.framework/Versions/4/Headers -I/opt/local/libexec/qt4/Library/Frameworks/QtGui.framework/Versions/4/Headers -I/opt/local/libexec/qt4/Library/Frameworks/QtCore.framework/Versions/4/Headers -I/opt/local/libexec/qt4/include -I../../generated/librecad/ui -F/opt/local/libexec/qt4/Library/Frameworks -F/opt/local/libexec/qt4/lib -o ../../generated/librecad/obj/rs_actioninterface.o lib/actions/rs_actioninterface.cpp
clang: error: unknown argument: '-fext-numeric-literals'
make[2]: *** [../../generated/librecad/obj/rs_actioninterface.o] Error 1
```

### Steps to reproduce or sample file

Build LibreCAD

### Operating System and LibreCAD version info

macOS 10.12.4
Xcode 8.3.2
clang 802.0.42 (Apple LLVM 8.1.0)
LibreCAD afb182a531228ec6c6f5b26e4bb44a829a965405

r-a-v-a-s, 2017-05-25:
In `src.pro`
```
*-g++ {
    QMAKE_CXXFLAGS += -fext-numeric-literals
}
```

You could try removing that;
although, it's written to be used only with g++ and not clang...

Maybe you need to run CLEAN.sh ?

kenrector, 2019-02-25:
How come this still happens a year and 9 months later?

sand1024, 2026-09-26:
outdated

closed 2026-09-26 by sand1024