LibreCAD/LibreCAD #894: LibreCAD 2.1.3 not available for Ubuntu

ManfredSteiner, 2017-05-26:
Please offer an stable LibreCAD 2.1.3 Debian-Packet for Ubuntu. One argument to use this software is the multi-platform ability. **Stable versions of the same version should be available on all platforms.**

The situation now:
* 2.1.3: LibreCAD for Windows
* 2.0.9: LibreCAD for Ubuntu LTS 16.04 (Xenial) - this is the most important version
* 2.1.2: LibreCAD for Ubuntu 17.04 (Zesty)
* 2.0.8: [ppa stable][1] - but no support for Ubuntu 16.10 and 17.04
* 2.1.1: [ppa daily][2] - but no support for newest Ubuntu 16.10 and 17.04

It seems, that the ppa is not longer supported.
Please add some information how to build LibreCAD directly from sources.

[1]: https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-stable
[2]: https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily

r-a-v-a-s, 2017-05-27:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

r-a-v-a-s, 2017-05-31:
This is a project maintained by a few volunteers.

Very few projects (in general) produce releases for all platforms.

Currently there is only one maintainer that produces releases (for Windows).

Github issues are things that can be closed by changing this repository.

You can use to forum for further discussion.
http://forum.librecad.org

lordofbikes, 2017-06-02:
Concerning the official Debian/Ubuntu distributions, this is a matter of the Debian/Ubuntu package maintainers. But it is more than once noted on the [Ubuntu package page][1] that it is not recommend to contact the maintainers directly.
There are already requests for official backports, e.g. in 2015 on the [Debian Maintainers mailinglist][2] without success.

I've updated the [daily ppa][3] with packages for 17.04 and 17.10, a 16.10 package was available already.
The 2.1.1+20170602.438 package contains the latest commits on master branch. It is named 2.1.1 as this is the common base with the stable version 2.1.3.

I will see what I can do for the [stable ppa][4] next.

[1]: https://packages.ubuntu.com/zesty/amd64/librecad
[2]: https://lists.alioth.debian.org/pipermail/debian-science-maintainers/2015-December/035849.html
[3]: https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily
[4]: https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-stable

ManfredSteiner, 2017-06-03:
@lordofbikes, Thanks for your effort.
Building from source was successful (thanks to @r-a-v-a-s ), and I was able to debianize this version (2.2.2) for Ubuntu 16.04, solving my request at the moment. In future, of cause, I would prefer to work with packages from official Ubuntu package page or ppa. So I hope you will be able to update packages to same version on all OS platforms.

bufalo1973, 2017-10-05:
@ManfredSteiner why don't you create a ppa for it? I don't know how to do it but I don't know either how to build a deb package :-)

ManfredSteiner, 2017-10-06:
Because there are repositories available. I hope, the maintainers will take my summarization to heart and do "their job" more duly. Normally it's not more work, you have only to organize it proper one time. Building a package can be a hard job, especially if you don't know the dependencies in detail. That job should be done by the software engineers of a project.

bkht, 2017-12-27:
Hi Manfred, is your build available for download?
I'm looking for a 2.1.3 version on Linux Mint 18.1 Serena.
Kind regards,
Jack.

ManfredSteiner, 2017-12-28:
@bkht, I recommend to build the binaries yourself, if you cannot find the desired version in the regualr software center (see link above). The debian packages for version 2.2.0, I build myself some months ago are available on this [link](http://www.htl-mechatronik.at/ubuntu-htl/pool/partner/librecad_2.2.0-alpha-187-gdb449e8_amd64.deb).

fa201, 2023-08-09:
Should we close this issue now that we have AppImage for Ubuntu ?