LibreCAD/LibreCAD #900: 2.2.0-alpha: Close document not working

serpien73, 2017-07-27:
When you close a document, the app should not still think the document is open.

To reproduce:

1. Open LibreCad. The default "Unnamed document" is active, as expected
2. Close the document via File/Close or "X" on the document window
3. The layer list for the document is still visible but greyed-out/disabled
4. The "Close" option is still enabled under the File menu

Version: 2.2.0-alpha
Compiler: GNU GCC 5.3.0
Compiled on: Jul 27 2017
Qt Version: 5.9.1
Boost Version: 1.53.0
System: Windows 7 SP 1 (6.1)

ashokgowtham, 2017-07-28:
Reproducible  👍

ashokgowtham, 2017-07-28:
Wen't through the code, I'll create a PR

ashokgowtham, 2017-07-28:
Regarding the "Close" option, It seems that there are other options that need to be disabled too:
Save, Save As, Export, Print, etc.

fa201, 2020-04-14:
Issue is still valid.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

sand1024, 2026-09-26:
reported for too old version

closed 2026-09-26 by sand1024