LibreCAD/LibreCAD #904: GUI freezes while autosave

djstkg, 2017-08-14:
### Expected behavior
Autosave should not hinder normal operation. Should run in background.

### Observed behavior
While active, autosave freezes UI. Editing .dxf file, autosave in #xxxxx.dxf

### Steps to reproduce or sample file
Just activate autosave and put save interval to 1 min.

### Operating System and LibreCAD version info
LibreCAD
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7

davemc50, 2017-12-08:
Even worse, the auto save feature isn't documented.  

The file name and location of auto save files is not clear. The behavior of multiple auto-save version  files is not mentioned, if it exists at all. 

On Linux the save file might be (drawing file name)~    ie, with a tilde suffix, in the same folder as the drawing.  There again, this might just be the prior manual save of the file.

Given that Librecad does crash from time to time, having the auto save feature actually working, documented and accessible would be of benefit to user retention.

Package: librecad
Architecture: amd64
Version: 2.1.2-1
Origin: Ubuntu

sand1024, 2026-09-26:
otdated issue

closed 2026-09-26 by sand1024