LibreCAD/LibreCAD #905: layer gets invisible while zooming

willybauss, 2017-08-27:
In my construction in have multiple layers (non-construction layers) containing similar data. One of them hides as soon as I zoom in. The others stay visible. Project is attached.

### Expected behavior
Layers should stay visible independent from zoom level.

### Observed behavior
Layer "Dielen senkrecht" hides as soon as I zoom in. All others stay visible, especially layer "Dielen waagerecht" which contains very similar data.
[terrassentreppe.zip](https://github.com/LibreCAD/LibreCAD/files/1254222/terrassentreppe.zip)

### Steps to reproduce or sample file
- unzip attached dxf file
- open it using LibreCAD
- zoom in and out while having a look on mentioned layer => it hides when zooming in

### Operating System and LibreCAD version info
Windows 10 Pro Version 1703
LibreCAD:
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows

Bruce56, 2017-09-21:
I'm experiencing a similar problem.  In my case, only portions of the layer disappear as I zoom in and reappear when I zoom out.  It's frustrating, because when I choose to zoom in, it's because I want to see a detail more closely, not have it become invisible!  I'm pretty sure this issue started with the latest version, as I was able to look at the same file with the previous version on my multiboot computer without this happening.  Thanks for the great software that keeps improving in spite of the occasional step backward!

willybauss, 2017-09-22:
Unfortunately it does not at all look like somebody's interested in fixing such issues. 26 days since I opened this thread, and no reaction so far :-( ...

Bruce56, 2017-09-28:
I found that if I select the items that disappear when zoomed in BEFORE zooming, then select attributes or properties, while actually making no changes to either, that I can get the disappearing to stop for those items.  You can select multiple items at once with this method.  I couldn't get this to work for call-out arrows with properties--only with attributes, so I suggest using attributes for everything.

This does not mean there isn't a bug, in my opinion.  It means there is a more or less tedious work-around.  

I have not yet tested to see if the problem stays away, or is just temporarily fixed using this method.  I've found that different layers misbehave at different times, so I'm guessing that it's not a permanent fix, but I'll report back if it seems to be permanent.

webmite, 2017-11-23:
FYI:  Tested the dxf file above using my development platform with no loss of visibility.
My platform is

> Version: 2.2.0-alpha-184-gafb182a
> Compiler: GNU GCC 5.4.0
> Compiled on: Nov 16 2017
> Qt Version: 5.5.1
> Boost Version: 1.58.0
> System: Ubuntu 16.04.3 LTS

I cannot currently check against the master branch because it will not compile without modification
of the project file on my platform.

So one thing to note is a newer gcc and an older boost library.

willybauss, 2017-11-23:
We do still not know why you can't see the issue, since you tested using a different version than the one I reported. Can you reproduce the issue using the version I mentioned above?

lordofbikes, 2017-11-23:
With release version 2.1.3 I reproduced your issue on Ubuntu successfully.
With the latest release the issue is solved. Thus it was fixed with one of the more than 200 commits since 2.1.3.

You can try the nightly builds for Windows. I believe the latest builds are as stable as the release version, don't be worried by the 'alpha' in the version tag.

There will be a new nightly build tomorrow as some pull requests were merged recently.

closed 2017-11-23 by lordofbikes