LibreCAD/LibreCAD #906: Snap on entity does not work in front of imported image

ant-t, 2017-09-13:
### Expected behavior

When a line is drawn with "snap on entity" the next drawing snaps to it.

### Observed behavior

Over an imported image "snap on entity" does not work.

### Steps to reproduce or sample file

Import any image and try snapping two object over it.

### Operating System and LibreCAD version info

Debian and the latest version from this repo.

fa201, 2021-01-24:
The issue is still there
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

natrius, 2021-04-19:
Yup, its the image. Was searching around and luckily finding your issue here. having the same problem. 

Version: 2.2.0-rc2
System: Windows 10

As a workaround i'm drawing useless lines to create an intersection where i can snap my orignal line.