LibreCAD/LibreCAD #908: build on qt get errors  "undefined reference to `vtable for RS_ActionEditPaste'"

parvizwpf, 2017-09-24:
hi.
when i build in qt get errros:
1 -\librecad\src\actions\rs_actioneditpaste.cpp:47: error: undefined reference to `vtable for RS_ActionEditPaste'

2- \libreCad\LibreCAD-master\librecad\src\actions\rs_actioneditpaste.h:39: error: undefined reference to `RS_ActionEditPaste::staticMetaObject'

alonso12, 2017-10-07:
Hello,

In Qt Creator, you may need shadow build turned on.

run clean project librecad.
run qmake
run rebuid librecad

https://stackoverflow.com/questions/1552069/undefined-reference-to-vtable-trying-to-compile-a-qt-project
Quote - "deleted the build folder, restarted Qt Creator and it worked"

These generally cure vtable probems.

Regards

sand1024, 2026-09-26:
old issue for outdated codebase

closed 2026-09-26 by sand1024