LibreCAD/LibreCAD #909: Import of images not working

kpanova, 2017-09-28:
I'm trying to import image (*.jpeg 
[image]
) and this is the result of import:
[image]
In version 2.0.9 import is working good.

## Operating System and LibreCAD version info

Operating System:
Windows 7 SP1
LibreCAD:
Version: 2.1.3

lordofbikes, 2017-09-28:
Draft mode activated?

http://forum.librecad.org/only-empty-boxes-frames-are-shown-for-image-text-no-line-thickness-td5706737.html

New tool bar icon in 2.1.3: [image]
Menu entry is same, View -> Draft

lordofbikes, 2017-10-11:
No response? So I assume draft mode did it.

If not, you're welcome to reopen this issue with new details and observations.

app4soft, 2023-07-19:
I has the same issue: after "Import Image" image not rendered in normal mode at all.

In "Draft" mode image border shown & its possible to select and modifi its attributes.

PNG image located in the same folder as DXF file.

@lordofbikes @dxli, reopen this issue.

[image]
[image]
[image]

lordofbikes, 2023-07-19:
Many thanks for reporting. I reopen it to keep your efforts.
Next time, please consider a new issue instead of reopen a 6 year old, unsolved issue.
A new issue, especially as it addresses the latest version, will be on top of the issues list and not hidden multiple pages down there.
Reference the old issue, to show you are aware of the closed old duplicate.

lordofbikes, 2023-07-19:
One idea that comes to my mind is an issue with path or filename, as it is mentioned in the forum.
So can you provide this information please.

Basically the image must be found and opened, otherwise I'd expect an error message somewhere.
Also the dimensions of the image must be red on import, because draft mode shows the wire frame.
Maybe it's some kind of escaping or OS path notation or code page issue?

lordofbikes, 2023-07-19:
Or you can relocate/rename the image to plain ASCII to prove this yourself.

app4soft, 2023-07-20:
Here is additional discovery:

- if create a new file and "Insert Image" (before file being saved to disk) image imported correctly and path look like: `../../../home/user/2.png`;
- if repeat "Insert Image" into same file after file saved to disk image imported invisible and path is just: `2.png`.

I'm using AppImage binary under Linux.

[image]
[image]