LibreCAD/LibreCAD #910: Feature request

dabeegmon, 2017-09-29:
<!-As a sorta hobbyist who finds that there always is a plethora of projects to do and never enough money to do what I need to do never mind what I want to do LibreCAD is helping me work more 'professionally'. I just can't afford spending even $1k usd and take the time to learn how to use it for somewhat commercial software and I am quite concerned about trusting my information and time to entities that show that they really only care about their profit. So I like LibreCAD! 

I think a comments page (or whatever) that I could attach to a drawing would be quite useful. Then I could write notes to myself (and anyone else if I'm sharing) to explain views and perspectives and reasoning. 

What say you?

(Not a coder but I would be interested in helping even if its to elucidate work flow.)

webmite, 2017-11-06:
If you want the comments to be embedded in a drawing you can create a layer specifically for comments and have a template that provides a editable text space for your comments.  Toggling visibility would mean it would only present itself when you have it 'enabled'.  Otherwise it is hidden and does not interfere with the drawing interactions.

If you are talking about an external document you are further ahead to simply use a standardized text-file or word document that has a filename that matches the drawing filename but has an alternate extension.  That should allow you to manage your files appropriately.

fa201, 2021-01-24:
I agree with @webmite for the comments. I also have a layer with text for explanation or things to do later.

But I think part of the request could lead to attach meta data with the DXF file which could be nice to have. There are similar things in office applications, Drasftsight, FreeCAD, etc.
Some fields like : name, company, contact info, version of drawing, licence.

sand1024, 2026-09-26:
supported by drawing metadata in Drawing Preferences

closed 2026-09-26 by sand1024