LibreCAD/LibreCAD #912: Image Importing

Kalthur, 2017-10-09:
When I insert a image into LibreCAD, I can't see it. I have searched the forum and I can't figure it out. I have a asbuilt I want to trace.

### Expected behavior
Insert Image jpeg
### Observed behavior
Cannot not see image in Librecad
### Steps to reproduce or sample file
Logged into Ubuntu 17.04 and Installed LibreCAD and does the same thing
Restarted WIndows 10 and made no difference.
Added a layer and then tried inserting the image and this did not work
Tried turning off draft and this made no difference
### Operating System and LibreCAD version info
Windows 10 and 
LibreCAD
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 

Also have this running on Ubuntu

lordofbikes, 2017-10-10:
Duplicate of #909

closed 2017-10-10 by lordofbikes

Kalthur, 2017-10-10:
Hi,

This has still not helped me, I can’t see the picture as was described in my request I have already tried the draft button!!

Thanks,

Closed #912<https://github.com/LibreCAD/LibreCAD/issues/912>.

lordofbikes, 2017-10-10:
I've seen, that you have checked draft mode, but it's still the same issue and it's easier for us to track this issue in one ticket.

The attached jpeg image there works for me, so we need more details to track it down.

You're welcome to add more information about your observations in #909. There was no response yet, if draft mode is the solution.

- Have you tried multiple images?
- Describe the import steps in more details.
- Give details about the jpeg image, pixel size, file size, color scheme (RGB or CMYK) or attach it if possible.
- Do you also see the wire frame after insert?
- Did you try auto zoom or disable and enable the layer again?
- Save the file after import, close and reopen it again, what happens?

Kalthur, 2017-10-10:
It is working now, not sure what I did but i see the picture

lordofbikes, 2017-10-11:
Nice to hear that and thanks for the response.