LibreCAD/LibreCAD #913: Current drawings preferences (dimensions, etc.) change when opening a previously saved file

jfirrantello, 2017-10-21:
### Expected behavior
Dimension settings do not change when opening a previously saved file.

### Observed behavior
I opened a previously saved DXF created in LibreCAD and found that the sizes of a part of the dimensions objects had changed significantly on re-opening. Values for items such as text height, arrow size, extension line offset, etc. had changed. I ''may'' have tracked it down to being a problem when "Inch" is set as the default unit for new files, but am not sure.

### Steps to reproduce or sample file
1. Clean LibreCAD Install
   1. Remove existing versions
   1. Remove local App data (Windows)
   1. Remove registry entries
   1. Download and install from website
1. Open LibreCAD
1. At opening splash screen, change default units for new drawings to "inches"
1. Options
   1. Drawing Preferences -> Paper -> "Letter"
   1. Dimensions -> General Scale -> "50"
1. Draw line using "2-points"
1. Dimension line using "Aligned"
1. Save file
1. Close LibreCAD
1. Open Sample File (using "Recent Files", not sure if that matters)

At this point, there are a variety of things that can appear wrong, and I cannot seem to get them to consistently appear. But, based on trying to reproduce the bug, it seems that the Drawing Preferences for the "Inch" drawing default are being overwritten by the Drawing Preferences for "None" or "Millimeter" drawing default. See image produced during attempting to reproduce issue [(link)](https://imgur.com/a/JwDnR).

### Operating System and LibreCAD version info
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7

waynepiekarski, 2019-06-24:
I am experiencing the same problem right now with LibreCAD from Debian 2.1.3-1.2 in June 2019. Did you find a solution to this?

jfirrantello, 2019-07-09:
> I am experiencing the same problem right now with LibreCAD from Debian 2.1.3-1.2 in June 2019. Did you find a solution to this?

I haven't really pursued it further, but have not found a solution.