LibreCAD/LibreCAD #914: cannot find some files, while LibreCAD2.1.3 is compiled by VS2015 and Qt5.7.1

liguosheng, 2017-10-23:
some files cann't be found,eg:

ui\forms\qg_cadtoolbar.ui，
ui\forms\qg_cadtoolbararcs.ui，
ui\forms\qg_cadtoolbarcircles.u
ui\forms\qg_cadtoolbardim.ui，
ui\forms\qg_cadtoolbarellipses.
ui\forms\qg_cadtoolbarinfo.ui，
ui\forms\qg_cadtoolbarlines.ui，
ui\forms\qg_cadtoolbarmain.ui，
ui\forms\qg_cadtoolbarmodify.ui
ui\forms\qg_cadtoolbarpoints.h，
ui\forms\qg_cadtoolbarpoints.ui
ui\forms\qg_cadtoolbarpolylines
ui\forms\qg_cadtoolbarselect.ui
ui\forms\qg_cadtoolbarsplines.u
ui\qg_actionfactory.h

lordofbikes, 2017-10-23:
Our Windows executables are built with MinGW and GNU toolchain. I think there is no active developer using MSVC. The MSVC specific files are VS 2010 and obsolete.

shawncurry, 2019-05-11:
> Our Windows executables are built with MinGW and GNU toolchain. I think there is no active developer using MSVC. The MSVC specific files are VS 2010 and obsolete.

There is now.  😎

fa201, 2020-04-13:
@lordofbikes 
Should we close this issue ?

sand1024, 2026-09-26:
outdated issues for old codebase

closed 2026-09-26 by sand1024