LibreCAD/LibreCAD #926: Crash when loading librecad (same version)-generated dxf

JohnCC330, 2017-11-18:
### Expected behavior
Normal load from a just-generated simple drawing

### Observed behavior
Crash on load (just a single arc!)

### Steps to reproduce or sample file

[curve.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/1484638/curve.dxf.gz)

### Operating System and LibreCAD version info
OS Linux Slackware current
LibreCAD 2.2.0-alpha-98-g7a4b5bb8
dxfrw 0.6.3

JohnCC330, 2017-11-18:
Compiler: GNU GCC 7.1.0
Compiled on: Nov 17 2017
Qt Version: 5.9.2
Boost Version: 1.63.0

Output on crash:

QG_LayerWidget::update: nullptr lastLayer
QXcbConnection: XCB error: 3 (BadWindow), sequence: 2777, resource id: 11612919, major code: 40 (TranslateCoords), minor code: 0
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.

QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
Segmentation fault
------------

JohnCC330, 2017-11-18:
gdb crash backtrace:
```
#0  0x000000000050adb1 in RS_FilterDXFRW::addHeader (this=0x2170d30, 
    data=0x7fffffffc618) at lib/filters/rs_filterdxfrw.cpp:1275
#1  0x00007ffff7af20cf in dxfRW::processHeader (this=this@entry=0x7fffffffc5e0)
    at libdxfrw.cpp:1868
#2  0x00007ffff7b0a935 in dxfRW::processDxf (this=this@entry=0x7fffffffc5e0)
    at libdxfrw.cpp:1835
#3  0x00007ffff7b0af09 in dxfRW::read (this=this@entry=0x7fffffffc5e0, 
    interface_=interface_@entry=0x2170d38, ext=ext@entry=true)
    at libdxfrw.cpp:99
#4  0x000000000050a7d8 in RS_FilterDXFRW::fileImport (this=0x2170d30, g=..., 
    file=..., type=RS2::FormatDXFRW) at lib/filters/rs_filterdxfrw.cpp:149
#5  0x0000000000503395 in RS_FileIO::fileImport (this=<optimized out>, 
    graphic=..., file=..., type=type@entry=RS2::FormatDXFRW)
    at lib/fileio/rs_fileio.cpp:76
#6  0x00000000004a83f6 in RS_Graphic::open (this=0x216dea0, filename=..., 
    type=RS2::FormatDXFRW) at lib/engine/rs_graphic.cpp:523
#7  0x00000000006ebae8 in QC_MDIWindow::slotFileOpen (
    this=this@entry=0x1117cf0, fileName=..., type=type@entry=RS2::FormatDXFRW)
    at main/qc_mdiwindow.cpp:355
#8  0x00000000006dffca in QC_ApplicationWindow::slotFileOpen (
    this=this@entry=0x7fffffffda60, fileName=..., type=RS2::FormatDXFRW)
    at main/qc_applicationwindow.cpp:1445
#9  0x00000000006e29b6 in QC_ApplicationWindow::slotFileOpen (
    this=0x7fffffffda60) at main/qc_applicationwindow.cpp:1338
#10 0x0000000000711a2f in QC_ApplicationWindow::qt_static_metacall (
    _o=0x7fffffffda60, _c=<optimized out>, _id=<optimized out>, 
    _a=0x7fffffffcba0)
    at ../../generated/librecad/moc/moc_qc_applicationwindow.cpp:375
#11 0x00007ffff61b2b0a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib64/libQt5Core.so.5
#12 0x00007ffff6ed8d72 in QAction::triggered(bool) ()
   from /usr/lib64/libQt5Widgets.so.5
#13 0x00007ffff6edb20c in QAction::activate(QAction::ActionEvent) ()
   from /usr/lib64/libQt5Widgets.so.5
#14 0x00007ffff7049ccc in ?? () from /usr/lib64/libQt5Widgets.so.5
#15 0x00007ffff7050d5f in ?? () from /usr/lib64/libQt5Widgets.so.5
#16 0x00007ffff7051c13 in QMenu::mouseReleaseEvent(QMouseEvent*) ()
   from /usr/lib64/libQt5Widgets.so.5
#17 0x00007ffff6f1ab88 in QWidget::event(QEvent*) ()
   from /usr/lib64/libQt5Widgets.so.5
#18 0x00007ffff7053e0b in QMenu::event(QEvent*) ()
   from /usr/lib64/libQt5Widgets.so.5
#19 0x00007ffff6edeeec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib64/libQt5Widgets.so.5
#20 0x00007ffff6ee67bf in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib64/libQt5Widgets.so.5
#21 0x00007ffff61855d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib64/libQt5Core.so.5
#22 0x00007ffff6ee580f in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /usr/lib64/libQt5Widgets.so.5
#23 0x00007ffff6f34b32 in ?? () from /usr/lib64/libQt5Widgets.so.5
#24 0x00007ffff6f37231 in ?? () from /usr/lib64/libQt5Widgets.so.5
#25 0x00007ffff6edeeec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib64/libQt5Widgets.so.5
#26 0x00007ffff6ee6264 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib64/libQt5Widgets.so.5
#27 0x00007ffff61855d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib64/libQt5Core.so.5
#28 0x00007ffff66e5c0b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib64/libQt5Gui.so.5
#29 0x00007ffff66e7555 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/lib64/libQt5Gui.so.5
#30 0x00007ffff66c0e1b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Gui.so.5
#31 0x00007fffeecf97e0 in ?? ()
   from /usr/lib64/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5
#32 0x00007ffff1c4214d in g_main_dispatch (context=0x7fffe40016f0)
    at gmain.c:3234
#33 g_main_context_dispatch (context=context@entry=0x7fffe40016f0)
    at gmain.c:3887
#34 0x00007ffff1c423f8 in g_main_context_iterate (
    context=context@entry=0x7fffe40016f0, block=block@entry=1, 
    dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3960
#35 0x00007ffff1c4249c in g_main_context_iteration (context=0x7fffe40016f0, 
    may_block=1) at gmain.c:4021
#36 0x00007ffff61db87f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#37 0x00007ffff6183bea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
#38 0x00007ffff618c474 in QCoreApplication::exec() ()
   from /usr/lib64/libQt5Core.so.5
#39 0x0000000000456aba in main (argc=<optimized out>, argv=<optimized out>)
    at main/main.cpp:315

```

JohnCC330, 2017-11-18:
This seems an error in the `libdxfrw` library - fairly near the start of the import file procedure.

Anything I can do to help?

JohnCC330, 2017-11-19:
I think this might be of interest. This strace shows the segfault happening just at the end:

```
...
write(5, "\1\0\0\0\0\0\0\0", 8)         = 8
recvmsg(6, 0x7ffd68b82760, 0)           = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=7, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=
read(5, "\1\0\0\0\0\0\0\0", 16)         = 8
poll([{fd=3, events=POLLIN|POLLOUT}], 1, 4294967295) = 1 ([{fd=3, revents=POLLOUT}])
writev(3, [{"+\0\1\0", 4}], 1)          = 4
futex(0xd4c2e8, FUTEX_WAKE_PRIVATE, 1)  = 0
futex(0xd4c2e8, FUTEX_WAKE_PRIVATE, 1)  = 0
poll([{fd=3, events=POLLIN|POLLOUT}], 1, 4294967295) = 1 ([{fd=3, revents=POLLOUT}])
writev(3, [{";\3\5\0\21\0\200\5\0\0\0\0\213\4\265\0\20\1\327\0\202\3\n\0\r\0\200\5\21\0\200\5\236\5N\3\213\4\265\0
recvmsg(6, 0x7ffd68b82760, 0)           = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=7, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=
write(5, "\1\0\0\0\0\0\0\0", 8)         = 8
open("/home/jcoppens//curve.dxf", O_RDONLY)       = 29
read(29, "999\ndxfrw 0.6.3\n  0\nSECTION\n  2\nHEADER\n  9\n$ACADVER\n  1\nAC1021\n  9\n$DWGCODEPAGE\n  3\nANSI_12
close(29)                               = 0
open("/home/jcoppens/curve.dxf", O_RDONLY)       = 29
read(29, "999\ndxfrw 0.6.3\n  0\nSECTION\n  2\nHEADER\n  9\n$ACADVER\n  1\nAC1021\n  9\n$DWGCODEPAGE\n  3\nANSI_12
--- SIGSEGV (Segmentation fault) @ 0 (0) ---
+++ killed by SIGSEGV +++

```

The raw values for the read() pointers are:

```
write(5, "\1\0\0\0\0\0\0\0", 8)         = 8
open("/home/jcoppens/curve.dxf", O_RDONLY)       = 29
read(0x1d, 0x2391150, 0x1fff)           = 0x1fff
close(29)                               = 0
open("/home/jcoppens/curve.dxf", O_RDONLY)       = 29
read(0x1d, 0x2391150, 0x1fff)           = 0x1fff
--- SIGSEGV (Segmentation fault) @ 0 (0) ---
+++ killed by SIGSEGV +++
```
Weird things (at least for my - very limited - knowledge:

  - I can't find any mmap which reserves anything like the size of the file (15915 bytes)
  - Why is the file opened and read twice?
  - Why is only 8192 bytes out of the 15916 read?
  - Why can't I find any mmap which reserves memory address at 0x2391150?

JohnCC330, 2017-11-19:
I've just recompiled LC with all -O0's so it's easier to follow the source code. The segfault is still there, but I have an extra line in the backtrace:

```
#0  0x00007ffff55702f0 in std::string::c_str() const () from /usr/lib64/libstdc++.so.6
#1  0x0000000000535af5 in RS_FilterDXFRW::addHeader (this=0x21cf880, data=0x7fffffffbf98)
    at lib/filters/rs_filterdxfrw.cpp:1275
#2  0x00007ffff7af20cf in dxfRW::processHeader (this=this@entry=0x7fffffffbf60) at libdxfrw.cpp:1868
#3  0x00007ffff7b0a935 in dxfRW::processDxf (this=this@entry=0x7fffffffbf60) at libdxfrw.cpp:1835
#4  0x00007ffff7b0af09 in dxfRW::read (this=0x7fffffffbf60, interface_=0x21cf888, ext=<optimized out>)
    at libdxfrw.cpp:99
#5  0x000000000052e6bc in RS_FilterDXFRW::fileImport (this=0x21cf880, g=..., file=..., type=RS2::FormatDXFRW)
    at lib/filters/rs_filterdxfrw.cpp:150

```

JohnCC330, 2017-11-19:
**SOLVED**. Thanks to Feragon on IRC, the culprit was the stand-alone libdxfrw 0.6.3 library from Sourceforge. There seems to be an incompatibility with Librecad (>2.0)
So, if you're having problems with a compiled Librecad, check if you have libdxfrw installed separately. If you do, I'd recommend removing it and recompiling, so LibreCAD uses its proper (internal) version.

**Note**:
If you're like me, and made a drawing with the bad version, if not too complicated, it is recoverable with a text editor (not for the faint of heart)

closed 2017-11-19 by JohnCC330

lordofbikes, 2017-11-19:
Have you tried to open another dxf file? Try to create another simple dxf with a 2 Point line and see what happens.

When the simple line dxf crashes too, there may be an issue with Qt or gcc version or maybe with the LibreCAD code base you use. 2.2.0-alpha-98-g7a4b5bb8 is about 130 commits behind current HEAD.

For me curve.dxf opens without issues. I use Ubuntu 16.04, Qt 5.5.1 and gcc 5.4.0.
My LibreCAD code base is latest master, which is 2.2.0-alpha-233-g0715edc.

What is suspicious with your dxf is the starting angle of the arc. In line 2676, after dxf code 50, the starting angle is noted as 14896.9026734014°. This is 41 * 360° + 136.9026734014°.
With my LibreCAD version I see an arc reaching from 136.9026734014° to 0°.

JohnCC330, 2017-11-19:
Hi lordofbikes,

I tried many dxf files, even looked up minimalistic dxfs on the net (like Paul Bourke's one).

As you can see above, with the help of Feragon, the problem was diagnosed as being a problem of the standalone version of libdxfrw. It seems that version wrote x, y, and z data wrongly (probably read them incorrectly too).

After removing the external 0.6.3 libdxfrw, and recompiling LC, all seems well.