LibreCAD/LibreCAD #930: LibreCAD source code compile problem

qium06, 2017-11-27:
My OS is win7. I use Qt Creator 4.4.1(based on Qt5.9.3). When compiling, I met with the following error.

\LibreCAD\librecad\src\lib\math\rs_math.cpp:305: error: no matching function for call to 'mu::Parser::DefineConst(const char [3], double)' 
         p.DefineConst("pi",M_PI); 
                                ^ 

\LibreCAD\librecad\src\lib\math\rs_math.cpp:306: error: no matching function for call to 'mu::Parser::SetExpr(std::__cxx11::string)' 
         p.SetExpr(expr_copy.toStdString()); 
                                          ^ 

\LibreCAD\librecad\src\lib\math\rs_math.cpp:312: error: no match for 'operator<<' (operand types are 'std::ostream {aka std::basic_ostream<char>}' and 'const string_type {aka const std::__cxx11::basic_string<wchar_t>}') 
         std::cout << e.GetMsg() << std::endl; 

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

qium06, 2017-11-28:
Has anyone successfully compile these source code before? It seems to be the problem of type conversion. The compiler can not automatically transfer type from "const char *" to "std::wstring" or from  "std::string" to "std::wstring" or from "std::wstring" to type suitable for "cout".

qium06, 2017-11-28:
These problems involve two cpp files. After I do changes to a few lines in these two files. The compilation is successful. I think there must be better ways to solve the problem between "std::string" and "std::wstring".

In the file  \LibreCAD\librecad\src\lib\math\rs_math.cpp:
rs_math.cpp
Line:305
p.DefineConst(L"pi",M_PI); //   <--  p.DefineConst("pi",M_PI);
Line:306
p.SetExpr(expr_copy.toStdWString());  //   <--p.SetExpr(expr_copy.toStdString())
Line:312
std::cout << QString::fromStdWString(e.GetMsg()).toStdString() << std::endl;  // <-- std::cout << e.GetMsg() << std::endl;

In the file   \libreCAD\plugins\plotequation\plot.cpp:
Line:60~64
            p.DefineConst(L"pi",M_PI);                   //<-- p.DefineConst("pi",M_PI);
            p.DefineConst(L"e",M_E);                     //<-- p.DefineConst("e",M_E);
            p.DefineVar(L"x", &equationVariable);        //<-- p.DefineVar("x", &equationVariable);
            p.DefineVar(L"t", &equationVariable);        //<-- p.DefineVar("t", &equationVariable);
            p.SetExpr(startValue.toStdWString());        //<-- p.SetExpr(startValue.toStdString());
Line:67
p.SetExpr(endValue.toStdWString());   //<-- p.SetExpr(endValue.toStdString());
Line:70
p.SetExpr(equation1.toStdWString());  //<-- p.SetExpr(equation1.toStdString());
Line:80
p.SetExpr(equation2.toStdWString());  //<-- p.SetExpr(equation2.toStdString());
Line:90
std::cout << QString::fromStdWString(e.GetMsg()).toStdString() << std::endl;  //<-- std::cout <<e.GetMsg() << std::endl;

rekado, 2017-11-29:
I packaged LibreCAD for GNU Guix, and we are successfully building it on GNU systems from source after only slightly patching the code to fix an error with Boost.

In `librecad/src/lib/math/rs_math.cpp` we include `#include <boost/serialization/array_wrapper.hpp>` before `#include <boost/numeric/ublas/matrix.hpp>`.  And in `librecad/src/lib/math/lc_quadratic.h` we do the same after `#include \"rs_vector.h\"`.

I would think that this is a Windows-only problem then.

qium06, 2017-11-30:
Thank you! @rekado

shawncurry, 2019-05-08:
Hi guys,

Windows 10; Qt 5.12.3; MinGW 7.3 32-bit.  I'm still seeing the issue referenced by the first post in the master branch as of today.

I have a proposed fix.  It's similar to qium06's proposal except that it should be safe for cross-platform and will support compiling in both UTF8 and UTF16.

I'm new to working on GitHub, this would be my very first pull request.  Normally at work, our process for submitting a bug fix goes like this:  create a branch for the fix, push to server, create pull request back into the milestone branch.  I'm currently stuck on step 2.

sand1024, 2026-09-26:
outdated issue for old codebase

closed 2026-09-26 by sand1024