LibreCAD/LibreCAD #936: Librecad freezes during zoom in with mouse wheel or zoom in button

davemc50, 2017-12-08:
Librecad freezes during zoom in

### Expected behavior
Roll mouse wheel, or click zoom in, program continues to function normally, zoom in is completed.

### Observed behavior
Program hard freezes, does not complete zoom. Terminate application required.

### Steps to reproduce or sample file
Roll mouse wheel forward, to zoom in. Intermittent, although once it has occurred the first time, it will continue to re-occur for every zoom, when program restarted thereafter.

### Operating System and LibreCAD version info
Package: librecad
Architecture: amd64
Version: 2.1.2-1
Origin: Ubuntu 17.10
Note: Multiple displays (3) of differing sizes and resolutions

xyr, 2017-12-19:
Are you using splines in your drawing when this problem occurs?

mdesplenter, 2018-02-17:
I have the same problem. In my case, I noticed it is a problem with a dashed/dotted construction-layer line. On a certain level of zoom the application stops working.

If I turn off one of the two options it will zoom, the combination is failing.

OS Windows 10

lordofbikes, 2018-02-18:
The construction line issue is already documented in #915, which sadly is work in progress still. It is not clear if this cause @davemc50's issue too.

doak, 2018-03-17:
I would like to link to my posted information on SourceForce:
https://sourceforge.net/p/librecad/bugs/595/#aace

This could be related to already linked #915 as well, I guess.

doak, 2018-03-17:
I can confirm that changes of #915 (cherry-picked to Debian stretch version (v2.1.2+b1) fixed the issue for me.

jcoffland, 2021-11-03:
I still have this problem with the latest code in the master branch.  As soon as I try to zoom LibreCAD locks up and wont recover.

xyr, 2021-11-03:
Back in the day (4 years!), I located the spline problem to these two files:

librecad/src/lib/engine/lc_splinepoints.cpp
librecad/src/lib/engine/rs_line.cpp

where the loops got stuck.

I didn't have the competence to properly fix it but I just added a couple of "sentinel" counters that  would exit those problematic loops after 1000 iterations. It was good enough for me and I was able to continue working with splines but that's pretty much all I can say about that. Now, when I run a diff against the current master branch, I can see that those files haven't changed from when I made my own modifications and at least I am still able to compile LibreCAD with those mods.

[spline_hack.tar.gz](https://github.com/LibreCAD/LibreCAD/files/7469911/spline_hack.tar.gz)

jcoffland, 2021-11-05:
@xyr, thanks.  I tried your patch but it didn't solve my problem.  It's still getting stuck.

I see these errors in the terminal:

```
lib/math/rs_math.cpp : quarticSolver : line 669
Quartic Error:: Found one real root for cubic, but negative
```
and
```
RS_Math::pow: EDOM in pow
```

Both are repeated several times.

sand1024, 2026-09-26:
cant' reproduce, too old issue

closed 2026-09-26 by sand1024