LibreCAD/LibreCAD #939: Properties dialog too small for HDPI, let remember last resize would be a fix.

bkht, 2017-12-27:
The 'Properties' dialog is too small for HDPI, so I resize it by mouse, making it larger, so the content fits well.

### Expected behavior
After I (re)open the Properties dialog again, I expect the same larger size as set using the previous properties dialog, so the contents still fit.

### Observed behavior
After I (re)open the Properties dialog again, the size is again too small to show the contents, they do not fit. I have to resize it again to make it larger so the content fit again.

### Steps to reproduce or sample file
Menu -> Dimension->Properties on some object like a line or circle.

### Operating System and LibreCAD version info
System:    Host: GS72 Kernel: 4.4.0-59-generic x86_64 (64 bit) Desktop: Cinnamon 3.2.7
Distro: Linux Mint 18.1 Serena
LibreCad 2.0.9-2

maxcel2, 2018-11-29:
Having the same problem on my laptop, screen resolution 1366x768. Have to resize it every time it is used otherwise data content is not visible.

sand1024, 2026-09-26:
not relevant anymore, persistent positions of dialogs are supported

closed 2026-09-26 by sand1024