LibreCAD/LibreCAD #945: Printing does not work - drawing clipped

mru00, 2018-01-21:
### Expected behavior

The program prints what is shown in print preview.

### Observed behavior

only part of the drawing is printed, most of it is clipped. also, zoom seems to be off.

[image]

Problem is the same if I print to PDF, so it is easy to show my problem:
[image]

### Steps to reproduce or sample file
[lenkrad.zip](https://github.com/LibreCAD/LibreCAD/files/1650008/lenkrad.zip)

### Operating System and LibreCAD version info

https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/2.0/LibreCAD-Installer-2.0.11-15-g90c1b83.exe/download?use_mirror=iweb&r=https%3A%2F%2Fsourceforge.net%2Fprojects%2Flibrecad%2Ffiles%2FWindows%2FNightlyBuilds%2F2.0%2F&use_mirror=iweb

on Windows 10 64bit.

mru00, 2018-01-21:
Problem dissapeared after selecting a scaling value:

[image]

I think the default was something like 1.2:1, not fixed or so.

Still, print preview should match what will be printed.

fa201, 2020-04-12:
@lordofbikes : we can close this issue.

closed 2026-09-26 by sand1024