LibreCAD/LibreCAD #946: drawing a circle "tangential 2 circles with radius": missing intersection

mru00, 2018-01-22:
### Expected behavior

I draw 2 lines, then use the "tangential 2 circles, radius" tool to draw a circle between them. 

I use the "trim" tool to trim the two lines to end where the circle starts, using the "snap intersection" mode.

### Observed behavior

Only one intersection point is found. The other one is not detected by the "snap intersection" mode.

Screenshot: the lower intersection point is detected, the upper one not.
[image]

### Steps to reproduce or sample file

draw 2 lines, draw circle using "tangential 2 circles with radius", then try to trim both lines using the intersection tool.

### Operating System and LibreCAD version info

[image]
on windows 10 64bit.

lordofbikes, 2018-01-22:
I tried some similar drawings, but it worked for me. Maybe it depends on the radius or line angles.
Can you attach your file for reference?

KasperHeyndrickx, 2018-03-18:
[946.zip](https://github.com/LibreCAD/LibreCAD/files/1822531/946.zip)
I could reproduce it with the attached drawing. It's only possible to snap to the top intersection. I suspect the circle is not touching the bottom line due to rounding errors. 
Adding a tolerance in RS_Circle::createTan2() solves it, but a tolerance in the snapping mechanism might be better. 
```
    auto e0=circles[0]->offsetTwoSides(r - TOLERANCE);
    auto e1=circles[1]->offsetTwoSides(r - TOLERANCE);
```

MikePryadko, 2020-02-04:
I observe the same in ~30% of cases: sometimes it's ok, sometimes only one point detected, sometimes no one
Example file: [librecad_bug-946.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/4154027/librecad_bug-946.dxf.zip)
How it looks like:
[image]