LibreCAD/LibreCAD #949: Add tool to make dogbone corners

Harvie, 2018-02-15:
Tool to quickly make dogbones of given diameter the same way we make fillets would be super usefull when drawing plans for CNC milling.

[image]

rightleftgreg, 2018-12-15:
That would be super handy

lordofbikes, 2019-02-26:
More possible options for fillet tool:
[image]

This is not intended to be exhaustive.

runningopenloop, 2019-02-28:
For slightly more completeness, here is what I'll call dogbone X and dogbone Y for lack of knowledge of better names.  As you mention, there could be inverses of these also.
[image]

Harvie, 2019-02-28:
@runningopenloop what if the lines are not exactly in X/Y direction, but at an angle? :)

lordofbikes, 2019-02-28:
maybe this:
[image]
It's all about the rules, independent from situation.
Standard: fillet arc is tangential to both entities
Dogbone edge: fillet arc's center is on the angle bisector and the circumference hits the entities intersection
Dogbone center: fillet arc's center is on the entities intersection
Dogbone X/Y: fillet arc is orthogonal to one entity and tangential to the other

Probably some of these varying fillets are useless on sharp or obtuse angles, but that applies to other tools too. See 'The Inside Corner Problem' at https://makezine.com/2012/04/13/cnc-panel-joinery-notebook/ for useful examples.
There will also be cases where a solution is impossible because of math.

You're welcome to join our [Zulip conversation](https://librecad.zulipchat.com/#narrow/stream/122956-Codebase/topic/V2.20DogBone.20feature.20(.23949)) about this.

runningopenloop, 2019-03-11:
I've started working on this.  Making some good progress.
- Got the center & inverse case done; inverse cases are relatively simple to implement.
- Working on the dogbone 'edge' which is mostly done.
- The dogbone X and dogbone Y are the same case with a different first line selected.  Not started.
    - In the above example of dogbone X, the first clicked line was the horizontal.  Agree?
    - In the above example of dogbone Y, the first clicked line was the vertical.  Agree?

Harvie, 2019-03-11:
@runningopenloop Nice to hear that! Please also note that to make sense in CNC use cases, the dogbone should only go far enough to cover the tip of original corner, but not further.

Eg.:

[image]

lordofbikes, 2019-03-11:
> * The dogbone X and dogbone Y are the same case with a different first line selected. Not started. 
>   * In the above example of dogbone X, the first clicked line was the horizontal. Agree?
>   * In the above example of dogbone Y, the first clicked line was the vertical. Agree?
Agreed!
So the rule is, the fillet is tangential to the first entity and orthogonal to the second one.

lordofbikes, 2019-03-11:
I thought about the CNC aspect too, but nonetheless, we have to consider many other use cases too.
Some examples to visualize my thoughts:
[image]
I see issues on angles smaller than 90°, because then the arc becomes greater than 180° and hence the opening smaller than the fillet diameter. See **d)**, where **A** is smaller than **D**, same applies to **a)**.

**a)** and **d)** are the straight versions, an arc added and trimmed with the entities.
**b)** and **e)** are limited to semicircles
**c)** and **g)** follow @Harvie's advice, the semicircles touch the intersection and are connected to the entities by additional lines
**f)** is like **d)**, the arc is in same position, but the lower edge is cut by an additional line to beware the fillet diameter.

In think **b)** and **e)** are nonsense, but I can imagine, that the others are useful.
I only wary about **f)** and **g)**. Similar to **a)** and **c)** the center and so the calculations for **d)** and **f)** are the same. But in contrast to **g)**, solution **f)** does not reach the intersection point.

So the questions are:
* Which fillet types are really needed?
* How to implement the options to keep the interface reasonable?

Harvie, 2019-03-11:
> Which fillet types are really needed?

I guess for CNC we need **c)**, which is identical to **a)** for angles >= 90°

Sometimes it might be usefull to have **g)**, which is the same as "**dogbone x/y**" proposed by @runningopenloop

runningopenloop, 2019-03-11:
> @runningopenloop Nice to hear that! Please also note that to make sense in CNC use cases, the dogbone should only go far enough to cover the tip of original corner, but not further.
> 
> Point of having such dogbones is to mill the **smallest** pocket needed to fit the desired corner using round endmill.

That was my original plan, however it did occur to me in my sleep that perhaps if you were concerned about the accuracy of your machine, maybe you would want it to push into the corner some (probably small) specified offset to ensure the intersections of your two pieces.  I'm doing wood with a [MaslowCNC](https://maslowcnc.com), so I'm thinking that any error will simply result in slight deformation of the wood... but maybe it would be nice to have some ability to enter an 'offset' margin.

Harvie, 2019-03-11:
> but maybe it would be nice to have some ability to enter an 'offset' margin.

You will still need to handle the offsets in CAM software... You are offseting the diameter of endmill, so just enter tiny bit smaller endmill than you actualy have. Some CAM software like bCNC even has separate setting for "additional offset" in CAM operations. (BTW bCNC is optimized to work nicely with LibreCAD drawings)

runningopenloop, 2019-03-13:
Thanks for sharing about bCNC @Harvie.  I'm not too familiar with CAM, but wouldn't it be trouble to specify the smaller endmill for just the corner dogbone operations?  Seems like I'd rather setup a an offset so it makes my CAM life easier.  However I see your point that CAD should be what you want, and then setup what you need in the CAM given your machine (at least I think that is your point in part or in whole).  Currently I've been using MakerCAM, I'm hoping to find a better replacement.  bCNC looks good.
  - However if the point of CAM is in part for dealing with limitations of your equipment, wouldn't dogbones also be considered something that should be implemented in CAM application?

I did finish the dogbone for 90 degree and obtuse angles.  Acute angles are as @lordofbikes case (a) at the moment as I've not done the tangent lines down from the circle to the entities that make the intersection.

As for which options are of value.  **I think the challenge in part may be coming up with good names for (a) through (g)**.  However I'm not planning to implement them all.  I'm just implementing them as I build upon the code to get to my goal of (c) and (g).  In the code I've got some case statements at a few key spots in the RS_Modification::round function in an attempt to re-use as much common code as possible.  So supporting the different options in the code seems to be not so bad.  Further, it continues to be trivial to support the inverse operations as they simply reverse the arc, at least so far.  Do we keep these options as they may be of interest when creating drawings?  This looks sort of cool if you wanted to make some sign or something.

[image]

Note, I've also got @lordofbikes "center" and "inverse" case from Feb 26th post.

Harvie, 2019-03-13:
> wouldn't dogbones also be considered something that should be implemented in CAM application?

You are very true. And bCNC can automaticaly create dogbones ("overcuts") when doing offset ("profile"). But sometimes i need specific dogbone configuration, that might be different in each corner. In such case it's part of the design process rather than CAM process and i want to do it in CAD. But most of the time i let the CAD figure it out automaticaly for me...

runningopenloop, 2019-03-13:
@Harvie, I think you mean "CAM figure it out automatically for me..."

We'll see.  :) I'm a big fan of Python, but have too many projects that aren't done.  Probably because I don't just buy some fancy polished CAD/CAM program.  :)

My next thing I was considering was adding a box joint modification to create the box joint profile to allow connecting objects together...  So I can draw the shapes as I want in CAD, then last step modify in CAD to have box joints.  I figure this would be best done in CAD as you'll need to design how things piece together.  However I expect there may be a CAM side for optimizing how the joints fit together?  Maybe you have some insight as to how this should be done?  Perhaps not by doing raw box joint modification, but by allowing the user to define seams where different CAD objects will go together.  I did a quick search, but didn't see this feature request in LibreCAD.  Maybe time to open it.  However I think I'm going to favor the quick simple raw add box joints.

Does this feature exist already and I'm not aware?

Then I can make my dice tower in style.  :)

Harvie, 2019-03-13:
bCNC has box generator. But only for block/cube shapes. If you need arbitrary shape, you're on your own. I don't have much experience with designing such stuff. But i can tell you how i fit such boxes.

Design the finger joints for ideal fit (gaps are exactly same dimension as fingers). Then cut it and test how it fits. If it's too tight/loose, add some negative/positive offset to the profile operation in CAM. Then you will learn what is the best additional offset for given machine+material combo. 

For me it usualy comes out little bit on the tight side. And if i am doing just one piece, i just force the joint together and then rub it a little to loosen up.

Paradoxically you don't have to care much about machine calibration, since both fingers and gaps are affected the same.

runningopenloop, 2019-03-15:
Ok, I've got the dogbone for acute angles working minus one detail.
(For some reason my text shows in a box after adding it in librecad, is there some option to turn on rendering?  I see the text before I place it.  My non-dev version of librecad works as expected)

Also ignore the debug reference points consisting of other circles and lines.
[image]

Case A) radius is < max distance between the line entities. --> So no problem here.
Case B) radius is = max distance between the line entities. --> I expect no problem here.
Case C) radius is > max distance between the line entities.  --> You can see what has happened, since I draw the tangent lines down until they intersect with the two original entities, we actually draw back to where the original drawing started.
   1) So do I just leave it as is.  If the user enters a radius > max distance, they can clean it up, or perhaps they want this?  Or
   2) Do I clip and just draw little horizontal lines over as I've manually marked up in D.  Or
   3) Reject the operation.  (What is the proper way to reject this?)

Depending on clean-up after this, I have only the tangent dogbone left to do.

73,
Timothy

lordofbikes, 2019-03-15:
@runningopenloop that looks promising.
I assume, that you mean diameter not radius.
So I would deny C) and D) at all. Check the condition ```2 * radius <= maxDistance``` at the beginning of your calculations and simply return ```false ``` in ```round()``` if not. This creates no preview and signs that there is no solution for current parameters.
It is possible to change the radius in tool options for a valid solution or cancel the tool.

I've checked the current fillet implementation, this is how it works. I was thinking about a hint in the command history, but ```round()``` is called on each mouse move event and will flood the history then.
Also this applies to extreme acute angles or big tools compared to drawing extents, I think that this feature is not intended for that.
I think we have to trust in user intelligence here :wink:

We should also think about nearly parallel line cases. Imagine two lines building an angle of 1° or less. There will be an intersection, but far away from our entities.
Any suggestions from all how to limit this? 
What make most sense? 
Use a minimum angle parameter? What is viable, 1°, 5°, 10° or what else?
Or limit the maximum overcut, e.g. ```radius * 5```, ```radius * 10``` or so?

runningopenloop, 2019-03-27:
Hi,
Got busy with life and work.  I added code tonight to return false if radius is too big.  I expect this will all be wrapped up in April.

Looks like there has been some code rot since I last worked on this.  Or rather I need to go back and ensure all my use cases are working fine.

So (1) re-validation of previous use cases, (2) add dogbone tangent, (3) clean-up, (4) check-in.

runningopenloop, 2019-03-27:
> I assume, that you mean diameter not radius.

Yes.

> We should also think about nearly parallel line cases. Imagine two lines building an angle of 1° or less. There will be an intersection, but far away from our entities.
> Any suggestions from all how to limit this?

Should I just return false (abort) if there is no intersection on actual drawn lines?  Similar to if the radius is too big case.

lordofbikes, 2019-03-27:
> Got busy with life and work.

Don't worry, that applies to most of us :wink: 
> Looks like there has been some code rot since I last worked on this.

What do you mean with that? The latest merges in master branch? You should work on a separate branch, then master commits don't affect your work and conflicts will be avoided.
Don't hesitate to ask if you need help with git.
>Should I just return false (abort) if there is no intersection on actual drawn lines?

Yes, but the question is when to abort and how to implement the limiting condition. Add an user adjustable option or hard code it?
This is only to avoid nearly infinite extension lines in rare cases.
I did some measuring starting with your example _A_, it's about 12° and the extension becomes ca. 8 times the radius. I think this is already an edge case.
For an angle of 1° between the lines, the extension is more than 110 time the radius. I think this is far away from a practical use case and should be save be hard coded. So I suggest to check for ```angle < 1°``` and ```return false;``` when true.

>(For some reason my text shows in a box after adding it in librecad, is there some option to turn on rendering? I see the text before I place it. My non-dev version of librecad works as expected)

Maybe you have _draft_ mode activated (Ctrl+D), then text and images are replaced by bounding boxes to speed up rendering.

Harvie, 2019-03-27:
> For an angle of 1° between the lines, the extension is more than 110 time the radius

Maybe bit exotic, but that still sounds like legitimate case for me. Given that user aplies dogbone manualy to each corner. I wouldn't like to introduce some unexpected behavior... Better leave it to the user, he can always choose not to apply on that sharp corner.  I think it might make sense for cases where the lines are something like 0.0001° parallel. In such case it should not create dogbone at all. Other than that it will only confuse the users...

runningopenloop, 2019-03-30:
> What do you mean with that? The latest merges in master branch? 

No, all my code.  :)  Things weren't working as I remember them working.  I think it had to do with adding support for the added in lines on acute angles.

> For an angle of 1° between the lines, the extension is more than 110 time the radius. I think this is far away from a practical use case and should be save be hard coded.

Just because it is, doesn't mean it is right - but the existing code for bevel doesn't seem to give regard for angle.  For example:
Angle: -0.1268° or 359.8732°
[image]

I repeated with same results for...
Angle: -0.0674° or 359.9326°

I don't think it's so bad that we need to add code to deal with...

Here's what happens on the original bevel code with parallel lines:
[image]
Just seems to extend toward the origin.  Better than a segfault or something.

> Maybe you have draft mode activated (Ctrl+D), then text and images are replaced by bounding boxes to speed up rendering.

Yes - thanks, that was it.

3kmch, 2019-09-25:
Hi @runningopenloop 

While searching for dogbone functions I stumbled over this issue. How about it? Do you plan to release it soon? That would be very handy. :)

Thanks for your work
Kurt